mshr-h/vscode-verilog-hdl-support #27: iverilog doesn't lint on .v file open

aviadb, 2018-08-01:
opening a verilog file
does not enables the linting automatically

aviadb, 2018-08-02:
1)C:\msys64\mingw64\bin
C:\msys64\home\aviadb\iverilog\iverilog_install\bin
are both in PATH variable

2. Yes
3. No

Raamakrishnan, 2018-08-02:
Which version of Windows and VS Code are you using?
Can you provide a minimal code that doesn't work with linting, along with your settings.json file?

Raamakrishnan, 2018-10-19:
I am sorry, you seem to be using msys2 to run iverilog. I haven't used msys2 myself. But I think you should be using the iverilog [windows binary](http://bleyer.org/icarus/) for it to work normally.

@mshr-h  Proposing to close this issue due to inactivity

gabrielebrajnikst, 2019-02-28:
@Raamakrishnan same issue here, I'm using the windows binaries.
1. iverilog is available in the PATH variable
2. iverilog is configured as the linter
3. Linting doesn't happen on saving the document

Windows 10.0.17763.316
VSCode 1.31.1

Raamakrishnan, 2019-03-04:
@gabrielebrajnikst can you try running iverilog through command line (from your workspace root) and let me know what the output is? Also, is it possible to reproduce this scenario in a small workspace and attach it here?
Thanks.

gabrielebrajnikst, 2019-03-05:
@Raamakrishnan  After some debugging, it turns out that even if I used the "official" installer, I had to manually put the following dlls in the same folder of iverilog executable:
- libgcc_s_seh-1.dll
- libstdc++-6.dll
- libwinpthread-1.dll
Now linting works.

Septyawan, 2019-05-20:
@Raamakrishnan sorry for asking dumb question, I just begin everything in Programming World. so how to check PATH variable in VsCode? I've exact same problem.

Raamakrishnan, 2019-07-02:
@Septyawan PATH variable is not something within VS Code. It is a OS level environment variable. The method to check it is different for different OS. Please google the method for you OS.

closed 2019-11-09 by mshr-h