mshr-h/vscode-verilog-hdl-support #30: there is a color mismatch bug

weiguoquan422, 2018-09-17:
when i type: 
module xadc(
    input i_clk_100m,
    input i_sig_p,
    input i_sig_n,
    input i_ref_p,
    input i_ref_n,
    input i_RESET
    );
the last sentence "input i_RESET"  display with wrong color.

mshr-h, 2018-09-20:
Could you upload a screenshot?

weiguoquan422, 2018-09-20:
[image]

weiguoquan422, 2018-09-20:
[image]

weiguoquan422, 2018-09-20:
just like this,Is it normal?

Raamakrishnan, 2018-11-22:
@brymatSF this seems to be a different issue and not the one @weiguoquan422 is reporting. I am able to reproduce your code and I myself have seen this many times. In OP, `input i_RESET` would be in pink colour and not red. So, that is different from what you report.

@mshr-h, in the issue @brymatSF is reporting, the port declaration is being detected as a module instantiation. Any idea how to fix this?

du33169, 2023-12-20:
Reproduced on v1.13.0. Any updates?