mshr-h/vscode-verilog-hdl-support #38: various extension support

atwok, 2018-10-12:
Can you add an option to just add additional file endings to associate with this extension?  I know I've see .v, .vh, .sv, .svh at work.  Sometimes others for various reasons.

Raamakrishnan, 2018-10-13:
You can use the `files.association` settings in vs code to add your custom extensions.
https://code.visualstudio.com/docs/getstarted/settings

kdurant, 2019-02-19:
This plugin can't highlight `.svh` file type

Raamakrishnan, 2019-03-03:
@kdurant I am repeating my statement here. Please add your extension to the file associations list that you want to be syntax highlighted. There are so many extensions that people use because the file extension do not matter to verilog/sv tools. So, it is a conscious choice to not include every single one of them.

Refer this https://code.visualstudio.com/docs/languages/overview#_adding-a-file-extension-to-a-language or the link I gave in earlier to set it up.

@mshr-h asking to close this issue. Should probably make some FAQ section in README and add this.

closed 2019-04-03 by mshr-h

mikeev261, 2019-05-22:
Can we please re-open this? I respectfully but completely disagree with @Raamakrishnan here. I've never seen a System Verilog header file named anything other than .svh. It seems to be a pretty well-established standard by now.