mshr-h/vscode-verilog-hdl-support #44: modelsim linting

kdurant, 2018-11-09:
[image]

how to set the argment？

kdurant, 2018-11-11:
I change  `command`
```typescript
let command: string = 'vlog -nologo -work $home/.cache/work \"' + doc.fileName +'\" ' + this.modelsimArgs; 
```
then message is `** Warning: (vlib-34) Library already exists at \"C:\\home/.cache/work\`.
But I use this command in terminal, it not display

Raamakrishnan, 2018-11-11:
I tried changing the command the way you did and it works just fine for me. I am not able to reproduce your warning. Even with that warning, I expect other error reporting to be working fine.

Yes, I agree that having an option for the work folder will be better. An idea would be to have a boolean that says "save the work folder in the workspace" (defaults to true). If it is true, insert the `-work work` argument automatically, else ask the user to insert the `-work` argument as part of the `verilog.linting.modelsim.arguments`. You can always give us a PR.

In `vlog`, if you set an argument multiple times, only the last value is taken. So, a workaround for this would be adding a `-work <custom path>` to `verilog.linting.modelsim.arguments` which will override the default path added by the linter.

kdurant, 2018-11-12:
@Raamakrishnan Sorry I won't use javascript, So....

kdurant, 2018-11-15:
@Raamakrishnan 

if only change the `command' in**../mshr-h.veriloghdl-0.3.6\src\linter\ModelsimLinter.js", it also generate a "work" folder in the current directory  

if change   the `command' in **../mshr-h.veriloghdl-0.3.6\out\src\linter\ModelsimLinter.js** , it display the warning ` Warning: (vlib-34) Library already exists at \"C:\\home/.cache/work\`

Raamakrishnan, 2018-11-15:
It seems like you are making changes to the installed extension rather than building it from source. So, making changes to the source ts files won't be reflected unless you build it again.
Besides, its a very common warning, I am sure it doesn't affect the workflow or prevents any other errors from popping up.

RLesma, 2019-07-10:
> I tried changing the command the way you did and it works just fine for me. I am not able to reproduce your warning. Even with that warning, I expect other error reporting to be working fine.
>
> Yes, [...]

Thank you, your extension is very useful and the -work trick worked perfectly. I have a question though, is there any way to know why the outline is shown empty? When I work in remote ssh the outline of Verilog files appears without problems.

Raamakrishnan, 2019-07-14:
@RLesma Do you have ctags installed in your location machine? On servers, ctags generally come pre-installed. That is probably why you can see outline only on remote ssh.

RLesma, 2019-07-14:
I have installed an extension called "CTags Support"by Jaydenlin, but it is possible I do not have ctags installed in my (Windows 10) machine. I'll check out if I can install it. Thanks!

RLesma, 2019-07-14:
@Raamakrishnan You were totally right, Ctags was not correctly installed. Again, thank you so much, excellent extension!

closed 2023-01-08 by mshr-h