mshr-h/vscode-verilog-hdl-support #45: instantiate module

markusdd, 2018-11-10:
Systemverilog extension already has instantiation functionality, but it sometimes does not work on verilog files and I only get offered the current file and submodules, but I want to choose from all of my workspace.

Is something like this planned for Verilog as well?

Raamakrishnan, 2018-11-11:
Yes it is one of the planned features. Please have a look at #25 

If you are interested, you can checkout the `sv` branch of this extension where this feature has already been added. It needs "universal ctags" to be installed for it to work.

markusdd, 2018-11-12:
Yes, I have seen that as I have this installed in parallel.
I can use the feature when I switch the language to Systemverilog, but it only shows files within the current files dir, not all possible files form the workspace.
I have a pretty large library set to work with, and the stuff I want to instantiate is almost always somewhere else.

How to fix this? Is there a search path to be set?

Raamakrishnan, 2018-11-13:
When you select "Instantiate Module" command, it shows the files in your current directory. The first option in the list is ".." (two dots, as in the linux way of saying "up"). If you select that it will take you one directory up. Then you can choose your folder and the file within it to get the module you want to instantiate.

[image]

markusdd, 2018-11-13:
Yes I noticed that, but it is very inefficient for me. I have to go up like 4 or 5 levels and then descend back into some other library (as I said, this is a larger project).

Would you consider adding an option to show all verilog modules in one list so that I can just search in it by typing a few chars and then directly select it? (You could take Sublime Text Systemverilog plugin as a reference, it works exactly this way and it was very convenient.)

Raamakrishnan, 2018-11-14:
Maybe in future