mshr-h/vscode-verilog-hdl-support #51: iverilog linting

ngoclinhle, 2018-12-20:
Hi,

All my module instantiation in my old code got error: "Unknown module type" after the update.
Simulation and synthesis by vcs and vivado still work.

Raamakrishnan, 2018-12-22:
Hey @ngoclinhle 
The iverilog error "Unknown module type" is thrown when you don't give iverilog all the files that contain your modules. Can you check how you have given the path in `` `include``s and how `verilog.linting.iverilog.runAtFileLocation` is configured? Have a look at the README for more info on this.

Can you try to recreate this scenario in a small test project/folder and attach it here?

aviadb, 2018-12-25:
Since last update, the linting is messed up.
Trying to lint closed files, non verilog files and many more bugs.
which pollute the "problems" tab

Raamakrishnan, 2018-12-25:
> non verilog files

this has been fixed in #53 . It will be out soon
> Trying to lint closed files

I think you are talking about the bug where the problems reported by "Rerun linter" command is not being removed on closing the file, then it is yet to be fixed
> many more bugs

Can you please file an issue?

anontruck, 2019-02-25:
Just to add, I'm seeing the same issues as @alexwhittemore, only I'm using Windows. I can build successfully using the `-y` flag to include my library directories, but including them to my linter arguments doesn't make the issue go away. Using `include`s doesn't help either. I'm sure I don't really understand Icarus either though.

I recently discovered that the most current windows build (available [here](http://bleyer.org/icarus/)) is slightly dated and does _not_ reflect the latest iverilog snapshot. There are at least a few compiler flags this version doesn't include, so perhaps that might be part of the problem? I doubt this is the same issue on macOS though.

Raamakrishnan, 2019-03-04:
@alexwhittemore,
As @anontruck was saying, you can use the `-y` flag of iverilog to tell the directory where all your files are present. From the iverilog man page:
> -y<libdir>
 Append the directory to the library module search path. When the compiler finds an undefined module, it looks in these directories for files with the right name

By "right name" they mean, the name of the file should be `<module name>.v`. Make sure all files are named like that (it is usually done that way, but just in case) and include the argument `-y <path to source files>/`. Your path can be absolute or relative to your workspace root

1. Adding `` `include`` is certainly one way of doing it. I understand the difficulty of doing it when it comes to IPs. Try the `-y` argument and let me know.
2. Yes that's a bug. While splitting the error message by ":", the last part is being left out. Can you make a separate bug so that it will be fixed properly? Thanks

@anontruck,
Thank you for mentioning about the `-y` flag. I completely forgot about that. I tried adding that to the linter arguments and it works fine for me.
1. The linter is executed from your workspace root (unless you have set `verilog.linting.iverilog.runAtFileLocation` to be `true`). So, if you are using relative paths in `-y` flag or `` `include``, it should be relative to your workspace root. Can you check those?
2. If your path contains spaces, they need to be enclosed within double quotes while passing it through `-y` flag in linter arguments. Can you check this too?

If none of the above solves your problem, try to replicate it in a small workspace and attach it here. So, I can take a look. Thanks.

anontruck, 2019-03-23:
@Raamakrishnan 
Sorry for dropping off... to follow up yes I was able to get it working! I was using the library include flag in my linting arguments before, but was defining the path incorrectly. All good now, thanks!

nahuel-cci, 2019-03-31:
As stated before adding `-y` flag solves the issue for `.v` files.
For SystemVerilog the flag turns into `-Y .sv -y <path to source files>`

shyeonn, 2022-03-11:
In vscode, 
[image]

but in terminal, i can compile this tb. 
what wrong in vscode

canseyman, 2022-04-20:
I have the same issue. If I include package, some modules are not recognized. Otherwise, I can't use the package.

cyjun99, 2022-10-24:
use like : iverilog -y .\ abc.v -o wave
"-y .\" will include all used files in the abc.v path

M6STAR, 2022-10-31:
The problem has not been solved, and the use of y parameter still cannot be solved, which is very uncomfortable.
The first figure shows that the module is missing,So I decided to use the include directive, but the terminal representation still couldn't find the file(Look at the second picture)
[image]
[image]
Next time I'll try relative paths,The results are still problematic,As shown in the third
[image]
I want to use y parameter, maybe I don't use it correctly, It still shows that the module cannot be found，please teach
[image]
[image]

YOlOKY, 2022-11-07:
I have the same problem with @M6STAR

iSlans, 2022-11-13:
@M6STAR 
if i'm not wrong, your `counter` module is defined in `test.v`...  but the `-y ./` will look for a `counter.v` file