mshr-h/vscode-verilog-hdl-support #54: How to prompt me for syntax errors.

NjtechPrinceling, 2018-12-24:
After installing this powerful plugin, my Windows and Linux VsCode don't have the fancy features like the ones you show (with syntax highlighting and auto-completion, but there is no function that can prompt me for syntax errors). I see yours. The sample diagram even has output variables for display. Where am I doing something wrong?   Thank you . and i‘m a Chinese Student , English is not good, please forgive me.

NjtechPrinceling, 2018-12-24:
How fast you can give my response ! and Thank you i will try now.

NjtechPrinceling, 2018-12-24:
And , What is Ctags ？ Should i install this and export it to my System environment  variable.

Raamakrishnan, 2018-12-24:
Ctags is a programming tool that generates an index (or tag) file of names found in source and header files of various programming languages [wiki](https://en.wikipedia.org/wiki/Ctags)
Yes, install it and have it in your `PATH` variable

NjtechPrinceling, 2018-12-26:
Hallo, I am back. I installed Ctags and imported the environment variables. It seems that my Code completion has changed. But I still don't understand how the plugin calls the emulator. My platform is Vivado. Which one should I choose in the four options? (It's like modelsim) And, don't I need to provide a specific path to the emulator? Or the plugin itself integrates the emulator.

Finally, is it true that the world alone will not configure this stuff? . . . . I cried.

NjtechPrinceling, 2018-12-27:
Sorry, I learned, thank you very much for your help, and I am honored to communicate with you through these networks!

closed 2026-01-15 by mshr-h