mshr-h/vscode-verilog-hdl-support #59: Modelsim Linter Not Working

CaiB, 2019-02-06:
Hi there,

I seem to be having some issues using the modelsim linter. It is installed, and I have put the installation directory (`C:\Program Files\IntelFPGA\modelsim_ase\win32aloem`) in my `PATH` variable (and running `vlib` from cmd does work), but at the top of all `*.sv` files, I get the following error:

> [modelsim] ** Error: (vlog-66) Execution of vlib.exe failed. Please check the error log for more details. [modelsim]

I'm not sure which error log it is talking about, otherwise I would have posted that as well. How do I go about resolving this?

Thank you!

gabrielebrajnikst, 2019-02-28:
Same issue here for all *.v files. Is there a possibility to "debug" the extension?

Raamakrishnan, 2019-03-02:
Can you check if your "work" library is present in your workspace root? Right now, the extension supports the design library only with name "work".
If it is not there, create it by running `vlib work` in your workspace root. If it is already present, can you try deleting it and creating it again?

The extension does not run `vlib`. It only runs `vlog` and expects the work library to be already present (a thing that should probably be added to the README). So, I am not sure I even understand the error being thrown here.

@gabrielebrajnikst Yes, you can. Clone the repo and start debugging. If this is your first vs code extension, you might find [this](https://code.visualstudio.com/api/get-started/your-first-extension) helpful. Any help is much appreciated.

CaiB, 2019-03-02:
Ah, that would do it. I had the next level up folder open as my root in VSCode, so the `work` folder was one level deeper instead of at the root. Using the subfolder as my root, it works fine.

I guess I have a feature request: look for the `work` folder not only in the root, but at the level of the `.sv` files instead, in order to allow multiple projects to be open in subfolders.

Thanks!

Raamakrishnan, 2019-03-03:
I am not sure if it would be useful to look for `work` folder inside the same directory as `.sv` files as these libraries tend to be stored separately in some location dedicated to store simulation related stuff (library cache, waveform files...)

I agree that forcing the `work` folder to be at workspace root is not optimal. It needs to be changed. Right now you can use [this](https://github.com/mshr-h/vscode-verilog-hdl-support/issues/44#issuecomment-437666250) trick to override your path to `work` folder
> In vlog, if you set an argument multiple times, only the last value is taken. So, a workaround for this would be adding a -work <custom path> to verilog.linting.modelsim.arguments which will override the default path added by the linter.

closed 2026-01-15 by mshr-h