mshr-h/vscode-verilog-hdl-support #62: Can't get CTags to work

soerensofke, 2019-03-03:
Hi,

I am trying to set up CTags to provide outlines, autocompletion, etc. But things will not come together.

I am using Windows 10, latest VSCode, CTags as of today and the most recent Verilog extension. CTags binaries are installed in C:\CTags, propper path is applied in VSCode extension settings.

Any ideas on how to debug?

Best
 Sören

Raamakrishnan, 2019-03-03:
Hi,
1. Are you able to run ctags from command line and is the output correct?
2. In the settings, `verilog.ctags.path` should actually point to the executable and not the folder in which it is located. I think this might be misunderstood from the README.

For example, 
Correct : `"verilog.ctags.path": "C:\CTags\ctags.exe"`
Incorrect : `"verilog.ctags.path": "C:\CTags"`

3. Can you try adding your ctags installation folder to `PATH` variable and setting `verilog.ctags.path` to `ctags`?

Thanks

soerensofke, 2019-03-03:
``
[image]

soerensofke, 2019-03-03:
Hi,

> Are you able to run ctags from command line and is the output correct?
I was only able to execute ctags on the command line after deleting the environment variable. See here for reference: https://github.com/universal-ctags/ctags/issues/1775

After, I corrected "verilog.ctags.path" to "C:\CTags\ctags.exe"

Now certain features of the plugins are working, e.g. outline, autocompletion.
Goto Definition and Peek Definition are still not working.

Any ideas on how to proceed?

Best, Sören

closed 2019-03-03 by soerensofke

Raamakrishnan, 2019-03-04:
Goto Definition and Peek Definition only works within a file for now. If it is not working within a file, can you attach the file where you are having trouble?

rliangcn, 2024-03-19:
I have the same question. I wonder why this issue, which seems unresolved, was closed.

Does this imply that the project is outdated and incompatible with newer versions of Visual Studio Code?