mshr-h/vscode-verilog-hdl-support #63: Syntax highlighting for casez and casex

danielkasza, 2019-03-07:
Syntax highlighting seems to be missing for `casez` and `casex`.

[image]

mshr-h, 2019-04-03:
Seems to be working in my env.

[image]

Code to reproduce

```systemverilog
module module_name(
    input  input_1;
    output output_1;
);

case (case_expression)
  case_item_1: begin
    case_statement_1a;
    case_statement_1b;
  end
  case_item_2 : case_statement_2;
  default     : case_statement_default;
endcase

casez (case_expression)
  case_item_1: begin
    case_statement_1a;
    case_statement_1b;
  end
  case_item_2 : case_statement_2;
  default     : case_statement_default;
endcase

casex (case_expression)
  case_item_1: begin
    case_statement_1a;
    case_statement_1b;
  end
  case_item_2 : case_statement_2;
  default     : case_statement_default;
endcase

endmodule
```

danielkasza, 2019-04-15:
I tried this again before updating. It was failing when I set the language to Verilog, but it worked in SystemVerilog mode.

The issue is now fixed.

closed 2019-04-15 by danielkasza