mshr-h/vscode-verilog-hdl-support #65: Modelsim linter conflit with SVN extension

Benz0X, 2019-03-18:
When using modelsim linter on a project under SVN (and only with the johnstoncode.svn-scm extension enabled), the linter throw an error on a non existing file.
Strangely, this doesn't impact the two other linters I can use (xvlog and verilator)

If i put syntax error in my file, the new error is shown, but the error on the non existing file is still thrown.

Workaround idea : check file existence before running the linter on the file ?

This is the error returned when copying from the PROBLEMS view : 

>
	"resource": "/FILE_DIRECTORY/jtag_common.v.svn",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": "modelsim",
	"severity": 8,
	"message": "** Error: (vlog-7) Failed to open design unit file \"/FILE_DIRECTORY/jtag_common.v.svn\" in read mode.",
	"source": "modelsim",
	"startLineNumber": 1,
	"startColumn": 1,
	"endLineNumber": 1,
	"endColumn": 1.7976931348623157e+308

Benz0X, 2019-04-11:
I started again from a clean install an wasn't able to reproduce the issue.
My guess is that there was a conflict with the svn extension, that created a temporary file $filename.svn and for some reason the linter tried to run on the file.

Anyway, the problem seems solved/non reproductible so you may close the issue.

On a side note, I tried to use the tmLanguage from Sublime text plugin systemverilog by Clams and the syntax highlight was greatly improved. (It can be found [here](https://bitbucket.org/Clams/sublimesystemverilog/src/cd79f5f9ca49d22224931046b7f35326345e4fd2/SystemVerilog.tmLanguage)). I didn't see any licence restriction so it might be interesting to use it in your plugin. Another great feature from this plugin is the verilog/systemverilog indenter/beautifier, but I'm not sure it can be easily ported.

Thanks for your answer

Raamakrishnan, 2019-04-11:
@Benz0X I am glad the issue is solved.

As of v1.0.3, that was released a few days ago, we have shifted to the tmLanguage of the sublime text plugin you mentioned. I will see what can be done about the beautifier.

Happy to answer. You can close the issue yourself :)

closed 2019-04-11 by Benz0X