mshr-h/vscode-verilog-hdl-support #66: Basic operation with ctags

fitch22, 2019-03-25:
Hello, please forgive me for asking such a basic question.  I have your extension installed along with the latest version of universal ctags.  I have added the path to the executable in the settings as described in issue #62.  I have created a tags file in the folder I want to open.  When I open the folder in vscode and open a file in that folder, none of the ctags functionality is present.  I assume I am missing a step, but am clueless as to what.  Can you help a complete idiot to get this to work?

By the way, I don't have any other ctags support installed, do I need any other extensions?  Right now all I have is this one.  Would really love to get the ctags to work.

Thanks for your help.

GaryBoone, 2019-03-31:
I'm having similar issues. On a Mac, I've confirmed that basic linting works. I have ctags running set up and running independently. But when I F12 over a module name, nothing happens but the vscode "spinner" equivalent (a bar moving across the top of the editor window.) I've also tried the `include and -y suggestions in #51.

BTW, do we have to run ctags ourselves, or does the extension do it? If the latter, where are the tags files stored by the extension? 

So in addition to fitch22's question, some advice on how to observe/confirm/debug the ctags activities would be appreciated.

Thanks in advance.

Raamakrishnan, 2019-04-09:
You don't have to create the tag file. The extension will run ctags automatically for the file you open and store its output in-memory. It works like tagbar extension for vim, if you are familiar with it.

@fitch22 you don't need any other extension to get this work. This extension will take care of everything. Which OS, VS Code version and extension version are you using?

@GaryBoone Right now, the only way to debug is to run the extension in VS Code extension development. There are no logs or such that you can view. I don't have access to a Mac to check this extension. Can you help us debug it?

You can try the following though:
* After you open the file, can you see the Outline pane being populated?
* Can you run ctags for your file from terminal the same way the extension runs it and check if you see the expected output?
This line tells you how the extension runs ctags https://github.com/mshr-h/vscode-verilog-hdl-support/blob/master/src/ctags.ts#L125. The command will be run at workspace root directory.

fitch22, 2019-04-09:
@Raamakrishnan I am using VSCode 1.33.0 and the extension is 1.0.3, but I have used several versions over the past couple weeks.  Nothing is working for me at all.  For example, if I right click on a module instance and select "go to definition", the sliding bar at the top starts going, and goes forever.
I have my ctags executable both in my path and in Verilog->Ctags: Path = "c:\users\bill\bin\ctags.exe".  I have tried with and without the double quotes, seems to make no difference.

Could it be because I have a large project?  I know ctags takes a while to run.

Raamakrishnan, 2019-04-10:
A large project should not have an impact because ctags is run only the file that is currently open. But, can you check with a very small workspace? If it doesn't work, can you attach the workspace here?

Can you also try the ideas that I mentioned in the previous comment?

> You can try the following though:
> * After you open the file, can you see the Outline pane being populated?
> * Can you run ctags for your file from terminal the same way the extension runs it and [...]

fitch22, 2019-04-10:
@Raamakrishnan OK, I think I have things working.  Couple things you might want to clarify as I had to experiment to find the right settings.

In the settings, I had the path to my ctags executable as "c:\users\bill\bin\ctags.exe".  I tried removing the ctags.exe from the end, no change.  What finally worked was to remove the double quotes.

So, not sure what to do about this.  I tried a test with a directory and 2 files.  One file had an instance of a module contained in the other file.  Right clicking on the instance displays no definition found.  Is there a way to navigate from file to file to follow a hierarchy?  This is a basic function of ctags in emacs and vi, and in fact the single most important reason I use it.  What am I doing wrong?

Thanks.

Raamakrishnan, 2019-04-10:
@fitch22 I am glad you got it working :)

Whatever you give in `verilog.ctags.path` will simply be placed in the terminal to run ctags. Since your path doesn't have any spaces, double quotes is not required. But then again, it should work with quotes. I might have to check on it.

Currently the extension supports only tags for one file (the currently open file) and not the entire workspace. That is something I am planning for the future.

GaryBoone, 2019-04-14:
@Raamakrishnan: 

• Yes, the outline is populated.

• Yes, running the ctags file according to the line of code given in your suggestion:

$ ctags -f - --fields=+K --sort=no --excmd=n "."

works fine, producing an output similar to the file created by "ctags ."

So it seems that it is working: I can hold down the command key (Mac) and click on a variable name to jump to the definition. 

What seems to not be working is definitions in other files. They're underlined with a red squiggle. I can add an "`include" line at the top to make the red squiggle go away, but command-clicking on the name doesn't jump to its definition in the other file.

hirooih, 2020-02-10:
> Currently the extension supports only tags for one file (the currently open file) and not the entire work space.

This should be documented in the README.md.
I've been using tags to look for symbols in other files.

> That is something I am planning for the future.

Yes, please.  We can create a tags-file by hand.

Thank you  for your great work.  Please continue it.

Pufferfish, 2020-12-09:
I installed ctags today and integrated it with the extension, I was sad to find that it only works for single files as mentioned above. Please make it a high priority task to add workspace support.
> Currently the extension supports only tags for one file (the currently open file) and not the entire workspace. That is something I am planning for the future.

Thanks!

hirooih, 2020-12-20:
> Currently the extension supports only tags for one file (the currently open file) and not the entire workspace. 

Try the latest Universal Ctags and [ctags companion extension](https://marketplace.visualstudio.com/items?itemName=gediminaszlatkus.ctags-companion) released today.  It merged my pull-request to support most of Universal-Ctags kinds including for Verilog/SystemVerilog. They works well with this extension.

And I've been working to improve the Universal-Ctags Verilog/SystemVerilog parser.  It works much better than before.  If you find any issues, please open issue tickets on Universal-Ctags GitHub.

closed 2023-07-20 by mshr-h