mshr-h/vscode-verilog-hdl-support #69: [Non feature request] Create issue template

kikd, 2019-04-08:
I was felt that this repository issues is not familiar for developer.
So that,  I propose create issue template to `.github/ISSUE_TEMPLATE/issue-template.md` for user create good issue easier.
 
I think bug report requires some information as below.

* Bug description

* Environment
Using OS, VS Code version, color themes, lint tools, etc.
I think color theme is important information for reproduce syntax highlight issue .

* Reproduce step (or data)
Screenshot is easy to know what happens. 
But, that is cannot "copy and paste" to editor, it makes harder reproduce. 

* Expected behavior

* Actual behavior

* Additional
For example, attach screenshot here.

--- 

Maybe, some reporter will need feature request template.

mshr-h, 2019-04-08:
Sounds good to me.

mshr-h, 2019-04-08:
Just added issue template.
Give us feedback so that we can make template better.

kikd, 2019-04-08:
LGTM
Thank you.

closed 2019-04-08 by kikd