mshr-h/vscode-verilog-hdl-support #70: Verilator lint support on Windows

jweng2, 2019-05-14:
I'm trying to run verilator on Win10 using a precompiled binary which has been added to PATH. From the terminal I can run verilator manually and get acceptable output, however when setting the appropriate options to lint using verilator the extension will not report any issues with invalid code. I'm not sure if this is because verilator support is entirely disabled, I notice that the compatibility table indicates that verilator is unavailable (possible owing to its linux-only release), but should the VSCode extension still try to run verilator on Windows when commanded to run a lint check?

terriblefire, 2019-06-26:
I was not able to get a modern ventilator to build on windows. The older binaries have incompatible options. Is the binary up to date?

jweng2, 2019-06-26:
So I am using an older version (3.922, 3-17-2018) built using the [toolchain here](https://github.com/FPGAwars/toolchain-verilator/releases).