mshr-h/vscode-verilog-hdl-support #71: [BUG]input, output keywords not colored in systemverilog

oaktwig, 2019-05-31:
As shown in picture below, the key words input and output not colored in .sv file.
[image]

kikd, 2019-06-03:
Hi.

I tried to reproduce at version 1.0.3 and "Dracura" theme,  but issue is not reproduced.

[image]

Would you mind tell us to information below?
* Extension version
* Using color theme

```
module one_bit(a, b, z);
	input   a, b;
    output  z;

    assign z = a + b;

endmodule

module four_bit();

    logic [3:0] z;
    logic [3:0] a,b;
    logic c;
    bit d;

    // do something

endmodule
```

Regards.

Raamakrishnan, 2019-07-02:
I agree with @kikd. I am unable to reproduce it as well. Looks like just a theme issue. Can you try using the "Inspect TM scopes" and post the details? Refer [here](https://code.visualstudio.com/api/language-extensions/syntax-highlight-guide#scope-inspector)

closed 2019-11-09 by mshr-h