mshr-h/vscode-verilog-hdl-support #73: A simple code completion bug

beginnerk, 2019-06-13:
[image]
When I use tab to complete the keyword "module",  Two cursors will appear.  
In detail, I write "mod" firstly, then I use tab to complete it. Then it will be
module(
    xxx
);
endmodule  xxx

xxx represent cursors. The two cursors have the same behavior. If I write codes, they will appear at the two position  concurrently.
my vscode run on Windows10

beginnerk, 2019-06-13:
[image]

jikjoo, 2019-06-18:
@beginnerk  
Hi, I think it might be a simple solution to input ESC.
Two cursor help me to write the module name in front of the module and behind comment at the same time.
It is not a bug, it's intended.

Raamakrishnan, 2019-07-02:
Yes, what @jikjoo said is true. Its a feature, not a bug 😄

closed 2019-11-09 by mshr-h