mshr-h/vscode-verilog-hdl-support #77: Development Plan 2

Raamakrishnan, 2019-07-18:
After a great first round, lets do it again?

These are the issues that were either left out unfinished in #25, requested by people or bugs
1. Auto Alignment (Format Document/Selection) - https://github.com/mshr-h/vscode-verilog-hdl-support/issues/25#issue-337197861
2. Module hierarchy - https://github.com/mshr-h/vscode-verilog-hdl-support/issues/25#issuecomment-403134096
3. Choose a different linter for a specific file - https://github.com/mshr-h/vscode-verilog-hdl-support/issues/25#issuecomment-427632438
4. Preprocessor support - #39 
5. Search for module across workspace and instantiate - #45 
6. Issue with "Rerun linter" - #55 
7. location of work library for modelsim - #59 

And these are the new features that I would like to propose:
8. File list

mahsoommoosa42, 2019-07-27:
I had a suggestion w.r.t verilator support on Windows via WSL. I had a glimpse at the code and I do feel it's certainly possible. we'll just have to add some checks to see if verilator is supported via WSL or MinGW. Perhaps keep two linter options, verilator(wsl) or verilator(MinGW)

yqszxx, 2020-03-24:
> I had a suggestion w.r.t verilator support on Windows via WSL. I had a glimpse at the code and I do feel it's certainly possible. we'll just have to add some checks to see if verilator is supported via WSL or MinGW. Perhaps keep two linter options, verilator(wsl) or verilator(MinGW)

I've checked the availability of verilator running in WSL, it will build and pass it's regression tests. So any further plan for adding WSL support?

mahsoommoosa42, 2020-03-24:
Its not required. If we run VS Code with the Remote extensions. It'll automatically detect it. This issue needs to be closed.

silabs-MattG, 2024-03-29:
Another item someone [requested for the original development plan](https://github.com/mshr-h/vscode-verilog-hdl-support/issues/25#issuecomment-418140840) isn't listed on this page:

> A must have feature(s) is the capabilities of verilog-mode to Emacs.
> auto-inst and auto wire connections

closed 2026-01-16 by mshr-h