mshr-h/vscode-verilog-hdl-support #78: [Support] Xvlog linter does not seem to work

agehall, 2019-08-08:
I've just installed the extension and tried to set it up to use xvlog as the linter. I've added "F:\Xilinx\Vivado\2019.1\bin" (which is where my Vivado lives) to my path and from the terminal inside VSC I can run "xvlog" but regardless I don't see any feedback from the linter in my code.

I can't see any relevant output anywhere in the output windows from this extension, so I'm assuming it fails miserably when attempting to run the linter.

Is there any way of getting more logging from it to see what is going on? Any other ideas I can try?

achakraborty2591, 2019-09-03:
It does not work for me too. I am using Windows 10. All the necessary paths have been added.

L0wMax, 2020-01-16:
I am using windows 10 as well and it does not work for me either.

RailRanger, 2020-04-14:
Same issue here, I tried to add that path and it does not do anything.

alexkarnaukhov, 2021-01-11:
same problem...

rebel-hskim, 2023-02-11:
You need to check the output log by doing the followings.
1. adding `"verilog.logging.enabled": true` to `settings.json`
2. Open "OUTPUT" tab (Ctrl+K Ctrl+H)
3. Open any verilog source file and see if the log `[Info] xvlog lint requested` displayed.

closed 2025-12-26 by mshr-h