mshr-h/vscode-verilog-hdl-support #84: [BUG]cannot peak module definition in another file

xiaoas, 2019-11-01:
**Describe the bug**
I've got universal-ctags installed, When I ctrl+click a module defined in another file, or reference a module defined in another file, the editor won't take me there / give me a peak of the module's definition.
This is a feature request (sorry if I posted it in the wrong way)

**Environment (please complete the following information):**
 - OS: Windows 10 1909 build 18363.449
 - VS Code version 1.39.2
 - Extension version 1.0.3
 - iverilog, universal-ctags

**Steps to reproduce**
Steps to reproduce the behavior:
1. in work folder, create a file named FOO.sv with module FOO
2. in work folder, create a file named USE_FOO.sv with module USE_FOO
3. reference module FOO within USE_FOO
4. hover your mouse over text "FOO"
5. right click text FOO, select "go to definition" / "peek definition"

**Expected behavior**
The editor should appear as if FOO is defined with the same file of module USE_FOO. in step 3/4/5, there shall be a peak of FOO's definition. 

**Actual behavior**
nothing really happens

**Additional context**
Sorry if this is impossible to implement, I am a student studying the language and the request might be unreasonable. I do think the feature can be helpful (as this is possible in other languages extensions)

guojiazhuxi, 2021-12-07:
I have the same problem

DavidZyy, 2023-02-02:
I have installed `universal-ctags` and the instantiate module works correctly.  But when trys to go to the definition of a module, it reports 'NO defination found for xxx'. I have no idea with the problem.

MahmoudKMaarouf, 2023-04-07:
Same problem. I am using command line arguments for Icarus Verilog -Y .sv -y "<path>"

cheerryio, 2023-05-06:
Same problem! Looks like there isn't any progress yet...✋

DavidZyy, 2023-05-06:
I recommand another extension：https://github.com/Nitcloud/Digital-IDE

MahmoudKMaarouf, 2023-05-06:
> I recommand another extension：https://github.com/Nitcloud/Digital-IDE
Thanks for the suggestion.

closed 2023-07-20 by mshr-h