mshr-h/vscode-verilog-hdl-support #88: [BUG] Highlighting color looking like incorrect

deweiwork, 2019-12-28:
**Describe the bug**

`https://i.imgur.com/fJvePXZ.png`

Above picture, red circle is correct I wished.

But when I typesetting, as following blue circle of picture.
`https://i.imgur.com/62A41vw.png`

The font color change to red, 
I think that font color should not be changed as red.

**Environment (please complete the following information):**
 - OS: [e.g. Windows 10 1809 build 17763.379]
 - VS Code version [1.41.1]
 - Extension version [1.0.4]
 - color themes [Atom one dark]

cebarobot, 2020-12-19:
I have this problem, too. I think this is same as #30.

I have done some test, which shows that this `input xxx` was matched to `Tags` or `entity.name.tag`...

cebarobot, 2020-12-19:
Ok, I found the problem...

If we write Verilog HDL code like 
```
module asdf (
    input some_input
);
```
Then `input some_input` will be matched to an `instantiation_patterns` in [syntaxes/verilog.tmLanguage#L156](syntaxes/verilog.tmLanguage#L156), whose pattern string is `^\s*([a-zA-Z][a-zA-Z0-9_]*)\s+([a-zA-Z][a-zA-Z0-9_]*)(?<!begin|if)\s*(?=\(|$)`. That is the reason why there will be incorrect highlighting color.

Maybe we can change the pattern string to `^\s*([a-zA-Z][a-zA-Z0-9_]*)\s+([a-zA-Z][a-zA-Z0-9_]*)(?<!begin|if)\s*(?=\()` which deleted the last `|$`. However, I am not sure weather there is any side effects.