mshr-h/vscode-verilog-hdl-support #89: [BUG] Linter fails with "Cannot read property 'logger' of undefined"

gtsiam, 2020-01-11:
**Bug description**
Linting of a verilog file produces the error _Cannot read property 'logger' of undefined_.
As far as I can tell this isn't limited to modelsim, but I only have modelsim installed.

**Environment**
 - OS: Windows 10, build 17763
 - VS Code version: 1.41.1
 - Extension version 1.0.4
- Modelsim: 10.5b (Starter edition) - Revision: 2016.10

**Steps to reproduce**
Steps to reproduce the behavior:
1. Open a simple verilog file with VS Code. (for me, the file was in the current vs code workspace).
2. Hit Ctrl+Shift+P
3. Run _Verilog: Rerun lint tool_
4. Select ModelSim and hit enter
5. Error appears as a popup within vscode on the bottom right.

As for the file, I tested both:
```verilog
module a;
endmoudk
```
and
```verilog
module a;
endmodule
```
with the same error.

**Log**
The only useful logs I could find where from the developer tools:
[image]

**Expected behavior**
I should see some linter output.

**Actual behavior**
I didn't see any linter output.

**Additional Context**
My username has greek characters in it, so maybe a unicode error? The project files however, are under C:\, so maybe that's unlikely.

This was broken by version 1.0.4. Looking at the changelog, I see changes to logger on 1.0.4, and it works fine on 1.0.3.

Brac24, 2020-01-23:
Did you install Icarus Verilog? I found that was my problem when I first ran it. Install Icarus Verilog and add it to your PATH system environment variable.

gtsiam, 2020-01-24:
This bug is for Modelsim (I was required to used modelsim in particular for my projects). But I would think it also applies to iverilog.

Even if it runs with iverilog, the bug remains. I dug a bit more into the code, but I know next to nothing about vscode extensions, so please - if one of the devs could take a look (@Raamakrishnan according to git blame, that's you):

The code I'm running (after transpilation I assume), is:
```js
    RunLintTool() {
        return __awaiter(this, void 0, void 0, function* () {
            // Check for language id
            let lang = vscode_1.window.activeTextEditor.document.languageId;
            if (vscode_1.window.activeTextEditor === undefined || (lang !== "verilog" && lang !== "systemverilog"))
                vscode_1.window.showErrorMessage("Verilog HDL: No document opened");
            // else if(window.activeTextEditor.document.languageId !== "verilog")
            // window.showErrorMessage("Verilog HDL: No Verilog document opened");
            else {
                // Show the available linters
                let linterStr = yield vscode_1.window.showQuickPick([
                    { label: "iverilog",
                        description: "Icarus Verilog",
                    },
                    { label: "xvlog",
                        description: "Vivado Logical Simulator"
                    },
                    { label: "modelsim",
                        description: "Modelsim"
                    },
                    { label: "verilator",
                        description: "Verilator"
                    }
                ], { matchOnDescription: true,
                    placeHolder: "Choose a linter to run",
                });
                if (linterStr === undefined)
                    return;
                // Create and run the linter with progress bar
                let tempLinter;
                switch (linterStr.label) {
                    case "iverilog":
                        tempLinter = new IcarusLinter_1.default(this.logger);
                        break;
                    case "xvlog":
                        tempLinter = new XvlogLinter_1.default(this.logger);
                        break;
                    case "modelsim":
                        tempLinter = new ModelsimLinter_1.default(this.logger);
                        break;
                    case "verilator":
                        tempLinter = new VerilatorLinter_1.default(this.logger);
                        break;
                    default:
                        return;
                }
                yield vscode_1.window.withProgress({
                    location: vscode_1.ProgressLocation.Notification,
                    title: "Verilog HDL: Running lint tool..."
                }, (progress, token) => __awaiter(this, void 0, void 0, function* () {
                    tempLinter.removeFileDiagnostics(vscode_1.window.activeTextEditor.document);
                    tempLinter.startLint(vscode_1.window.activeTextEditor.document);
                }));
            }
        });
    }
}
```
From out/src/LintManager.js

You can see `this` is overridden when the generator function is defined. Anyway, this may have been introduced by the transpilation - In short I doubt this is a missing tool problem.

Brac24, 2020-01-24:
Got it. As a sanity check you can run the simulator in a command prompt correct?

gtsiam, 2020-01-25:
Yes, it works fine: (I run `vlog -lint .\select8.v`).
And for reference, downgrading to (extension version) 1.0.3 works fine as well.

tarikgraba, 2020-01-31:
Same error when using Verilator on a Debian 10 machine.

The `verilator` command is in the PATH and  works as expected when called with `--lint-only` as argument.

Raamakrishnan, 2020-02-02:
The `Rerun lint tool` command has been broken for quite a while (#55). I wouldn't use it.
The file will be automatically linted on saving it. So, that should be good for most of the cases.

tarikgraba, 2020-02-02:
> The `Rerun lint tool` command has been broken for quite a while (#55). I wouldn't use it.
> The file will be automatically linted on saving it. So, that should be good for most of the cases.

I was testing the lint command because since the last update, the automatic lint on saving did not work any more.