mshr-h/vscode-verilog-hdl-support #102: [BUG] localparams added as params in module instance

markusdd, 2020-05-08:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment (please complete the following information):**
 - OS: macOS catalina
 - VS Code version 1.45
 - Extension version 1.0.6

**Steps to reproduce**
create a simple module using a localparam
instantiate this module using the instantiate module command

**Expected behavior**
localparams are not added to instantiation

**Actual behavior**
localparams are added to instantiation, which is syntactically and semantically wrong

markusdd, 2020-05-09:
[image]

The problem is most likely here. Constants also incorporante localparams und specparams.

The fix would be to quickly parse the actual string that ctags gives and check if the definition actually starts with 'parameter'

markusdd, 2020-05-10:
Parsing that out of ctags is currently tedious.
I ahve opened an upstream issue to provide a cleaner way of seperating defines, localparams, parameters and specparams:
https://github.com/universal-ctags/ctags/issues/2537

hirooih, 2023-07-11:
Note that Universal Ctags fixed the issue in Oct. 2020, https://github.com/universal-ctags/ctags/issues/2537#issuecomment-714516129.

Mightlaus, 2023-12-03:
Sorry, I still don't know how to correctly configure ctags to ignore `localparam` when instantiating.

I've tried adding `--fields-Verilog=+{parameter}` arguments, but it seems to result in opposite effect -- Only the `parameter` constants are being ignored, while all other `localparam` constants persist when instantiating.

This is my plugin setting:
[image]
This is the instantiation results:
[image]
---
**Environment:**:
Ctags for win: v6.0.0-x64
VScode for win: v1.84.2
Plugin: v1.13.0

hirooih, 2023-12-10:
@Mightlaus 

I added a new feature to Universal Ctags to fix this issue three years ago. I am disappointed to know that it has not been used.

Although I have little experience for typescript programming, I made a fix and sent a PR (#457).
As I wrote in the PR, the fix is not tested, because I don't have environment to test VS code extension.

I hope this helps this project.

RickyTino, 2024-08-12:
Hi, I am quite sure this untested fix has caused #479 , which occurs on v1.13.1 but not seen on v1.13.0. 
Until now ctags still recognizes all kind of params as 'constant' but not 'parameter’
[image]