mshr-h/vscode-verilog-hdl-support #103: Modelsim linter always gives 0 errors/warnings returned

RailRanger, 2020-05-10:
I have tried with modelsim student 10.4 and the intel 20.1 version. I am on Windows 10 x64.
No matter what I do, in the output of the extension I always get:

> [Info] Symbols Requested: file:///e%3A/Documents/verilog/work/test.sv
> > c:\ctags\ctags.exe -f - --fields=+K --sort=no --excmd=n "e:\Documents\verilog\work\test.sv"
> [Info] 0 errors/warnings returned
> [Info] 5 top-level symbols returned

However when I try to compile inside of modelsim it throws errors as expected.

danlomeli, 2020-05-11:
I'm having the same issue in Ubuntu console versus the debugger output

```console
vlog -lint axi_scheduler_v1_0.sim/axi_scheduler_tb.sv
Model Technology ModelSim - Intel FPGA Edition vlog 10.5b Compiler 2016.10 Oct  5 2016
Start time: 16:57:12 on May 10,2020
vlog -lint axi_scheduler_v1_0.sim/axi_scheduler_tb.sv 
-- Compiling module arbiter_tb
** Error: (vlog-13069) axi_scheduler_v1_0.sim/axi_scheduler_tb.sv(27): near "logic": syntax error, unexpected "SystemVerilog keyword 'logic'", expecting ';' or ','.
** Error: (vlog-13069) axi_scheduler_v1_0.sim/axi_scheduler_tb.sv(39): near "logic": syntax error, unexpected "SystemVerilog keyword 'logic'", expecting ';' or ','.
End time: 16:57:12 on May 10,2020, Elapsed time: 0:00:00
Errors: 2, Warnings: 0
```
debug output
```console
[Info] Symbols Requested: file:///home/daniel/playground/axi_scheduler/axi_scheduler_v1_0.sim/axi_scheduler_tb.sv
> /usr/local/bin/ctags -f - --fields=+K --sort=no --excmd=n "/home/daniel/playground/axi_scheduler/axi_scheduler_v1_0.sim/axi_scheduler_tb.sv"
[Info] 1 top-level symbols returned
[Info] modelsim lint requested
[Info] 0 errors/warnings returned
```

CaiB, 2020-11-12:
Has anyone found a solution for this issue? I'm seeing the same thing.

danlomeli, 2020-11-12:
No, I just went back to v1.0.3

CaiB, 2020-11-12:
Thank you. I can confirm downgrading the extension to 1.0.3 immediately fixed the issue.

CaiB, 2021-02-01:
Is there any plan to fix whatever is causing this issue?

Currently, linting breaks every time the extension gets updated until I downgrade to 1.0.3. Furthermore, if ModelSim has not been run (creating a `work` directory), linting also does not work. In both cases I just get no errors at all.

fadialzammar, 2021-03-31:
I'm not sure if this has to do with using QuestaSim instead of ModelSim, but I'm still having this issue even after downgrading to 1.0.3. If I record my screen to get a look at the command prompt before it disappears, I can even see that `vlog` ran properly after saving the file in VS Code and was able to find the error.
```
** Error: (vlog-13069) c:\Users\...top.v(20): near "input": syntax error, unexpected input, expecting ')'.
End time: 18:38:02 on Mar 30,2021, Elapsed time: 0:00:00
Errors: 1, Warnings: 0
```
Any ideas why it keeps reporting back no problems inside VS Code?

William-An, 2021-12-23:
I have also experienced the same issue on a remote machine via vscde-remote.

```
Extension version: 1.5.3
OS: Ubuntu 20.04
VSCode version: 1.63.1
Linter: modelsim
```

I took a look into the child process part in `src/linter/ModelsimLinter.js`, and it looks like it cannot found `vlog` and quit quietly without any message despite `vlog` is in my PATH. Replacing the command with `env PATH=$PATH:/path/to/modelsim/bin vlog -nologo -work` solve the issue.

MahmoudKMaarouf, 2023-04-19:
I have the same issue. 
I can have no path/an incorrect path for my modelsim linter and it will still report 0 errors and warnings.

This user here does not have this issue however:
https://www.youtube.com/watch?v=-DTGf3Z6v_o&ab_channel=RichBaird

So I am a bit puzzled by this situation.

danlomeli, 2023-04-19:
I was able to work around this. I had some time a few weeks ago to dig into the extension source and found the code that creates a local work dir is no longer able to do so. I created static work dir using vlib and pointed to it like this

vlib /home/<username>/work
code settings.json
...
"verilog.linting.modelsim.work": "/home/<username>/work"
...

MahmoudKMaarouf, 2023-04-20:
Thanks danlomeli for your help
But unfortunately, my situation still remains.
I have put a work folder in my home directory /usr2/mmaarouf/work yet the issue persists

With other simulators, I can view the command in the extension output window. Then I can copy and paste that command and try it out on my linux terminal. But for modelsim, it does not show this command. 

My path is pointing to Vlog and my work directory is shown as above. No sure what I am missing. Slang and icarus verilog works for me however.

jonpovey, 2024-06-14:
Thanks @danlomeli that fixed modelsim linting for me.
For the dev - it would be great if an error was reported in the Verilog module output for cases like this, rather than just `0 errors/warnings returned`

aqilc, 2025-01-23:
Downgrading the extension to 1.0.3 isn't available anymore, as the version is not available online. This is happening to me too currently, using the modelsim provided by Quartus. Is there any way to fix this?

aqilc, 2025-01-23:
UPDATE: Creating the work directory and updating the Work path in the Verilog extension worked! I hope this issue gets filed soon.