mshr-h/vscode-verilog-hdl-support #110: Instantiate Module

halfblood87, 2020-07-12:
verilog.instantiateModule

popup notifications:

Verilog-HDL/SystemVerilog:No modules found in the file

Ali-Flt, 2020-08-07:
I have the same issue and There is no syntax errors in my modules.
Those this feature only recognize a certain way of module coding ?

hunterzju, 2020-08-19:
I have the same issue:
macOS10.15.6+vscode insider+iverilog

Raamakrishnan, 2020-08-20:
Instantiate Module needs ctags to detect the modules present in the file. Please check if you have it installed

RyanLei-icer, 2025-02-12:
为什么在Linux版本的vscode的这个插件中添加ctags路径后还是报相同的error

[image]

[image]

[image]