mshr-h/vscode-verilog-hdl-support #120: [ENHANCEMENT] Please, publish your extension to open-vsx.org additionally to MS visualstudio marketplace.

jjchico, 2020-10-22:
Visualstudio marketplace is a great place to publish vs code extensions but [its use licence](https://github.com/microsoft/vscode/issues/31168) only allow it to be used with the non-free Visual Studio Code software built by Microsoft, but not with the open source vs code product or derivatives like VS Codium. Publishing your extension to open-vsx.org is a great way to reach a great number of users that respect licence terms and prefer to use the open-source vs code product instead of the non-free build published by MS.

open-vsx.org offer [detailed instructions](https://github.com/eclipse/openvsx/wiki/Publishing-Extensions) on how to publish to their repository, including an automatic github action.

Thanks.

jorge.

mohe2015, 2021-11-08:
Do you plan on doing that anytime soon? Would be really great as otherwise we can't use it with VSCodium

mshr-h, 2021-11-10:
I'll do that ASAP.

mshr-h, 2021-11-10:
Published to [Verilog-HDL/SystemVerilog/Bluespec SystemVerilog – Open VSX Registry](https://open-vsx.org/extension/mshr-h/veriloghdl).

mohe2015, 2021-11-10:
@mshr-h It's published now, thanks. Unfortunately it seems to not be compatible with 1.61.2

mshr-h, 2021-11-11:
@mohe2015 Fixed.

closed 2021-11-12 by mshr-h