mshr-h/vscode-verilog-hdl-support #122: [Enhancement] Linting: need xvlog argument support

QianfengClarkShen, 2020-11-19:
First, thank you for providing such a powerful tool!

My problem:
For linter, you support iverilog, modelsim, vivado xvlog, and verilator. However, all the linters except for xvlog have arguments support. If possible, could you help add that? Basically, I need to do "xvlog -L uvm" to enable uvm syntax check.

Thanks,
Clark

mshr-h, 2020-11-21:
Added xvlog argument support on v1.3.6.
Please update and see if it works.

QianfengClarkShen, 2020-11-22:
Thank you very much! It works like a charm.

alexkarnaukhov, 2021-01-11:
> Thank you very much! It works like a charm.

Hi! Does it really works like a charm for you? I cant see any output just like guys from this thread:
https://github.com/mshr-h/vscode-verilog-hdl-support/issues/78