mshr-h/vscode-verilog-hdl-support #123: [BUG] keyword "endcase" did not match "case" when highlighting

idlesilver, 2020-11-20:
**Describe the bug**
The keyword "case" and "endcase" do not match with each other in highlighting brackets. (Only the "case", other keywords like module, bracket, begin/end can be highlighted normally)

**Environment (please complete the following information):**
 - OS: Windows 10.0.19041
 - VS Code version 1.51.1
 - Extension version 1.3.5
 - HIGHTLIGHT EXTENSION: Bracket Pair Colorizer 2
 - color themes: monokai pro.

**Steps to reproduce**
Steps to reproduce the behavior:
copy code
```SystemVerilog
case(keycode)
    8'h1A : begin
                Ball_X_Motion_in = 0;
                Ball_Y_Motion_in = ~(Ball_Y_Step) + 1'b1;
    end
    8'h16 : begin
                Ball_X_Motion_in = 0;
                Ball_Y_Motion_in = Ball_Y_Step;
    end
    8'h04 : begin
                Ball_X_Motion_in = ~(Ball_X_Step) + 1'b1; 
                Ball_Y_Motion_in = 0;
    end
    8'h07 : begin
                Ball_X_Motion_in = Ball_X_Step;
                Ball_Y_Motion_in = 0;
    end
endcase   
```
See error
[image]

**Expected behavior**
Seen "case" and "endcase" as an pair of brackets when highlighting. 

**Actual behavior**
"case" can be seen as bracket, but not "endcase"

mshr-h, 2020-11-21:
Cannot reproduce the bug.
It's correctly working on my PC.

- OS: Windows 10.0.19041
- VS Code version 1.51.1
- Extension version 1.3.5
- color themes: monokai pro.

[image]

idlesilver, 2020-11-23:
This bug only appears when using the HIGHTLIGHT EXTENSION: **Bracket Pair Colorizer 2**
This extension allows matching brackets to be identified with colours. All other keywords matches each other except "case/endcase".

PanTongLin, 2021-04-28:
This is because "endcase" is common end bracket for "case", "casez", and "casex" in Verilog and Systemverilog.
Since "cases" and "cases" are rarely used, you can just remove them in "{EXTENSION_ROOT}/MSHR-H.VERILOGHDL-x.x.x/verilog.configuration.json" to fix this problem.
[reference](https://github.com/CoenraadS/Bracket-Pair-Colorizer-2/issues/328)