mshr-h/vscode-verilog-hdl-support #139: [BUG]iverilog linting doesn't work

doee-hc, 2021-02-19:
**Describe the bug**
Thank you for supporting such a powerful extension！
I have installed iverilog and add it into the PATH,it works well in "cmd" and "power shell"
Then I set iverilog as the linter from the VScode setting page.
It doesn't work when saving the .v file (no red underline and no output).
I have seen the similiar issuess, but these solutions doesn't work.
I have tried some old versions of this extension and iverilog,but the problem wasn't resolved.

**Environment (please complete the following information):**
 - OS:  Windows 10 1909 build 18363.1379
 - VS Code version 1.53.2
 - Extension version 1.4.2
 - Iverilog version v11-20201123-x64

yizhanchengyan, 2021-04-16:
被编辑的文本的路径不能有中文

kkwangyu, 2023-04-04:
The path of the edited text cannot contain Chinese and spaces。

furai, 2023-05-01:
Hey, I'm getting this bug in the current version of the extension. No linter works if the path to the file includes spaces. The fix would be to pass the path to the linter quoted.

mshr-h, 2023-06-02:
I guess it's fixed in v1.11.5.

canxin121, 2024-12-15:
Still not working. I have no chinese or space in the path.

CatalpaEel, 2025-05-15:
I just encountered the same problem and solved it. I have downloaded the Iverilog under the directory "Program File", which has a space in the path. So when I printed "iverilog demo.v" in the cmd, it returned an error. After I change the path of Iverilog, the linter seems to be working properly. I hope this will be helpful to you.