mshr-h/vscode-verilog-hdl-support #141: [Feature] Consider adding support for Google's verible linter and autoformatter

mithro, 2021-02-20:
It would be awesome if your code supported [Google's verible](https://github.com/google/verible). Verible is a suite of developer productivity tools for SystemVerilog code bases including;
 - Linter
 - Autoformatter
 - Kythe integration for indexing and crosslinking
 - [more](https://google.github.io/verible/README.html#systemverilog-developer-tools)

Prebuilt binaries are provided Linux systems and I'm sure they would love help making it work on [MacOS X](https://github.com/google/verible/issues/260) and [Windows too](https://github.com/google/verible/issues/307).

ilanbiala, 2021-03-25:
+1 specifically for the formatter, since it seems like this extension doesn't currently have formatting support. It seems like this also encompasses #126.

matthewbridges, 2023-07-07:
I think this issue is closed. Verible is supported

closed 2023-07-17 by mshr-h