mshr-h/vscode-verilog-hdl-support #147: Module last parameter syntax highlight error

sheep94lion, 2021-04-04:
The last parameter of a module in Verilog is shown in incorrect color:
[image]

alexisfrjp, 2021-05-09:
You should provide the file to reproduce the error. I've never seen this error.

fishcrap, 2021-07-30:
I do encounter the same problem. If keyword **wire** is omitted in the last line of a module definition, the parameter will be in wrong color.
[image]

DLehenbauer, 2022-09-02:
@alexisfrjp - The key to reproducing this bug is to put the closing ')' on the next line.

As you can see, this causes the last variable to be incorrectly highlighted as a keyword:

[image]

If you move the closing ')' to the same line as the last variable the highlighting is correct:

[image]

To reproduce, simply create an 'empty.v' file and paste the following:

```
module repro(
    input white,
    input blue
);
endmodule;
```