mshr-h/vscode-verilog-hdl-support #149: error: (vlog-7) Failed to open ini file \"modelsim.in"

13669283592, 2021-04-25:
When I want to use modelsim to check language, if I add the path of the Work, then there is a error. 

** Error: (vlog-7) Failed to open ini file "modelsim.ini" in read mode.

{
	"resource": "/c:/Users/lilei/Desktop/class/shift_reg_md.v",
	"owner": "_generated_diagnostic_collection_name_#1",
	"code": "modelsim",
	"severity": 8,
	"message": "** Error: (vlog-7) Failed to open ini file \"modelsim.ini\" in read mode.\r",
	"source": "modelsim",
	"startLineNumber": 1,
	"startColumn": 1,
	"endLineNumber": 1,
	"endColumn": 1.7976931348623157e+308
}

How can I solve this problem!

Noahyahh, 2024-10-09:
Hey,are u slove this problem? I find it in my vscode

xnaoica, 2025-01-22:
请问您解决这个问题了吗
我现在配置modelsim  也是出现这个问题，
每个.v文件第一行一直波浪线