mshr-h/vscode-verilog-hdl-support #153: Running the contributed command: 'verilog.lint' failed.

ScortYang, 2021-05-13:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment (please complete the following information):**
 - OS: Windows 10 20H2 build 19042.985
 - VS Code version 1.56.1
 - Extension version 1.4.3
 - xvlog

**Steps to reproduce**
Steps to reproduce the behavior:
1. Run command "Rerun lint tool"'
2. See error "Running the contributed command: 'verilog.lint' failed."

RomaGlock, 2021-05-13:
@ScortYang  Did this work in the previous version of the extension? I just started using the linter today)

ScortYang, 2021-05-13:
> @ScortYang Did this work in the previous version of the extension? I just started using the linter today)

Yes, it works fine in version 1.0.3 and earlier.