mshr-h/vscode-verilog-hdl-support #208: Variables cannot be completed automatically

hhucchenyixiao, 2021-12-19:
**Describe the bug**
Variables cannot be completed automatically

**Environment:**
 - VS Code version: 1.63.2 (user setup)
- Extension version v1.5.3
- OS: Windows_NT x64 10.0.22000

**Screenshot at failure**
[image]

**But if I don't save the file as .v, it can work normally**
[image]

Hola39e, 2021-12-29:
I met the same bug, and I changed the VSCode version back to 1.61, then the autocomplete works rightly.

hhucchenyixiao, 2021-12-30:
> I met the same bug, and I changed the VSCode version back to 1.61, then the autocomplete works rightly.

I now directly replace it with other plug-ins, such as TerosHDL or FPGA Develop Support.

markusdd, 2022-02-01:
I was able to fix this by setting word completion to current document instead of all documents.

Thsi seems to be a change in behavior in VS code.

The mentioned other extensions are not really usable, especially Teros HDL tries to impose its complete workflow onto you.

wytWangLei, 2022-04-19:
> 

so how to set word completion to current-document?

wytWangLei, 2022-04-19:
@markusdd so how to set word completion to current document?

markusdd, 2022-04-19:
does not work anymore also...

wytWangLei, 2022-04-19:
@markusdd I've just found that setting, but after saved as .v file the 'abc' intellisense still disappear permanently. So we can't autocomlete variables and other keywords like 'input' 'output' 'assign' ?

markusdd, 2022-04-19:
currently it seems like it...

wytWangLei, 2022-04-19:
@markusdd ok, but it's really confusing why some key words are still not supported as snippets...... is there any other verilog editors recommended on Win10?

ricwu, 2023-07-07:
[image]
this bug is still exists！！！