mshr-h/vscode-verilog-hdl-support #289: [BUG]

hyperupcall, 2022-06-06:
**Describe the bug**
Launching any extension with this as a dependency will clutter the debug console

**Environment (please complete the following information):**
 - OS: Linux
 - VS Code version 1.67.2
 - Extension version 1.67.2
 - color themes, lint tools, etc: Any

**Steps to reproduce**
Steps to reproduce the behavior:
1. Run an extension in debug mode (in the image below, i'm using [mine](https://github.com/hyperupcall/vscode-hyperupcall), but any should do)
2. See debug console

**Log**
```sh
mshr-h.veriloghdl is now active!
ctags manager configure
Invalid linter name.
Invalid language server name.
```

**Expected behavior**
No output

**Actual behavior**

[image]

**Additional context**
Unfortunately, out of my 150+ installed extensions, this one is the only one that manually prints information to the debug console. The only other output is from an unintended error from a different extension. Maybe it would be better to log informative output to a dedicated output window?

mshr-h, 2022-12-12:
It's not a bug but I understand it's annoying. Will fix.

mshr-h, 2022-12-20:
Should be fixed on v1.6.0.

closed 2022-12-21 by hyperupcall