mshr-h/vscode-verilog-hdl-support #311: [BUG] Displays a warning about no .svlint.toml on every startup

gussmith23, 2022-07-16:
**Describe the bug**

Every time I start VSCode or open a project, I get the warning about no .svlint.toml.

**Environment (please complete the following information):**
 - OS: MacOS 11.6.7
 - VS Code version 1.69.1
 - Extension version 1.5.4

**Steps to reproduce**

Open any project

**Expected behavior**

I guess it's technically correct to give the warning, but the warning is annoying in projects that have no Verilog!

mshr-h, 2022-11-12:
The message is coming from svls.
https://github.com/dalance/svls/blob/9c606d04e923841212368fdee0c71478889940a4/src/backend.rs#L270

I don't know how to disable it.

mshr-h, 2022-12-12:
Since v1.5.5, the extension won't be loaded until the supported file is opened.
It's not 100% fixed but I guess it's less annoying now.

closed 2022-12-28 by mshr-h