mshr-h/vscode-verilog-hdl-support #323: Use iverilog linter on a remote SSH file

angelo558, 2022-10-07:
Hi guys,
I am currently trying to have the graphic linter of iverilog using a remote SSH connection to a virtual desktop containing those files.
The SSH connetion is established using "Remote - SSH" extension of Microsoft, the iverilog package is installed under /pkg and I even added the allias iverilog to point to it directly in my bashrc file.

However, no matter what I try, it seems the linter is not working on my files.

Would you have any idea why and/or how to resolve this? 

Thanks in advance!

backbone-link, 2022-11-12:
Same issue here. Syntax errors not caught (i.e. missing `;`)

angelo558, 2022-11-14:
@backbone-link : could you please confirm that you managed to make iverilog work through a SSH connexion using "Remote - SSH" by Windows? and the only issue was the configuration settings?
If so, could you please share which settings you had to work on please?

Thanks!

Brillouin233, 2024-03-15:
@angelo558 :Did u solve this problem? Thank you very much!!