mshr-h/vscode-verilog-hdl-support #324: Couldn't start client SystemVerilog language server svls

ChongChengAC, 2022-10-13:
**Describe the bug**
Launching server using command svls failed, though `svls` is already in PATH

**Environment (please complete the following information):**
 - OS: Manjaro Linux
 - VS Code version 1.72.1
 - Extension version 1.5.4

**Steps to reproduce**
Steps to reproduce the behavior:
1. Install svls via `cargo install svls`
2. Make sure that svls is already in PATH: `svls --version`
3. Edit vscode Verilog configuration -> Verilog: Language Server to svls
4. Reopen vscode.

**Log**

[image]

**Expected behavior**
No errors.

**Additional context**
There's a similar issue https://github.com/mshr-h/vscode-verilog-hdl-support/issues/154#issue-891305794. But I installed svls by the way mentioned in its readme
[image]

mshr-h, 2022-11-11:
Released v1.5.7 with the updated config option language server.
You can specify the path to the svls binary in your config.
Check out the README for how to config it.

```
verilog.languageServer.enabled (Default: false )

verilog.languageServer.name (Default: svls )

verilog.languageServer.path (Default: svls )
```

ChongChengAC, 2022-11-11:
> Released v1.5.7 with the updated config option language server. You can specify the path to the svls binary in your config. Check out the README for how to config it.
>
> ``` [...]

It works after I fill `verilog.languageServer.path` with full path. Thanks!

closed 2022-11-11 by ChongChengAC