mshr-h/vscode-verilog-hdl-support #326: Stop showing "bsv only support one opened workspace now"

dominiksalvet, 2022-11-06:
**Describe the bug**
Every time I open the VSCode, the message `bsv only support one opened workspace now` pops up. I am using [multi-root workspaces](https://code.visualstudio.com/docs/editor/multi-root-workspaces) containing projects written in various languages. So it is quite annoying to dismiss this message every time I open VSCode, even though it may not affect my current project under work. Additionally, I do not understand the message itself -- please, what does it mean, what would I lose?

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04.1 LTS
 - VS Code version 1.73.0
 - Extension version 1.5.4

**Steps to reproduce**
Steps to reproduce the behavior:
1. Create multi-root workspace
2. Save workspace
3. Close VSCode
4. Open VSCode
5. Load workspace
6. The message pops up

**Expected behavior**
Nothing is shown -- bsv should either work in multi-root workspaces or be loaded into a single project from the workspace (e.g. of the currently opened file). Or e.g., loaded manually, or add an option to disable this message...

**Actual behavior**
This below happens on every VSCode open.

[image]

dominiksalvet, 2022-11-10:
UPDATE: Since #327 has been solved, this issue occurs only when I open a SystemVerilog/Verilog project for the first time in a VSCode session. Nevertheless, it is still present.

mshr-h, 2022-11-11:
Fixed on v1.5.6

dominiksalvet, 2022-11-11:
Hello, thank you! It seems that it works on v1.5.6. :+1:  However, on v1.5.7 I am getting the following error on loading the extension, which is probably unrelated but it may be related to the multi-root projects, IDK:

[image]

mshr-h, 2022-11-12:
@dominiksalvet 
Fixed on #334

dominiksalvet, 2022-11-12:
I confirm it is fixed on v.1.5.9. Okay, I am happy, thank you! :heart_eyes:

closed 2022-11-12 by dominiksalvet