mshr-h/vscode-verilog-hdl-support #327: Do not activate/load extension until required

dominiksalvet, 2022-11-06:
**Describe the bug**
The extension is activated on VSCode startup:

[image]

I personally use [multi-root workspaces](https://code.visualstudio.com/docs/editor/multi-root-workspaces) and I have projects written in various languages, so the extension is often activated unnecessarily and adds extra start time.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04.1 LTS
 - VS Code version 1.73.0
 - Extension version 1.5.4

**Steps to reproduce**
Steps to reproduce the behavior:
1. Install the extension
2. Close and open VSCode
3. Open extensions
4. See it activated and that took some time (e.g., 349 ms)

**Expected behavior**
It may be activated on some events. E.g. see the screenshot of Pylance config:

[image]

**Actual behavior**
It is activated on every VSCode startup.

mshr-h, 2022-11-10:
Because this extension is currently not adopted to Multi Root Workspace API.

mshr-h, 2022-11-10:
Ah, never mind.
`activationEvents` is incorrent. I'll fix it.

dominiksalvet, 2022-11-10:
Thank you! It works great. :heart:

PR #330 (2022-11-10): activate extension when supported file opened
Fix #327
closed 2022-11-10 by dominiksalvet