mshr-h/vscode-verilog-hdl-support #332: [BUG] verilog.languageServer is enabled by default

nfproc, 2022-11-11:
**Describe the bug**
A new language server feature is enabled by default in version 1.5.7.
As a result, an error popup appears that says the server cannot be launched.

**Environment**
 - OS: Windows 10 21H1 build 19043.2251
 - VS Code version: 1.73.1
 - Extension version: 1.5.7

**Steps to reproduce**
1. Update the extension to version 1.5.7
2. Make sure the user's `settings.json` file DOES NOT contain a line that disables the feature, i.e.:
>     "verilog.languageServer.enabled": false,
3. Open a supported file (.v, .sv, etc.)

(Step 2 also means a possible workaround is adding the above line.)

**Expected behavior**
No errors happen.

**Actual behavior**
An error popup appears.

**Additional context**
Perhaps the problem lies in the default value of `verilog.languageServer.enabled` in the `package.json` file.
It should be `false` rather than `"false"`.

mshr-h, 2022-11-11:
Thanks! Can you make the PR if possible?

nfproc, 2022-11-12:
Ok, I created a PR (#334).
BTW, this issue made me aware of a great feature of the new version. I will take some time to utilize the feature shortly.

commit 51b06704 (2022-11-12): Resolves #332 (#334)
closed 2022-11-12 by mshr-h