mshr-h/vscode-verilog-hdl-support #348: [BUG] `property` token breaks bracket colorization

djoffe, 2022-12-06:
**Describe the bug**
A fairly recent feature of vscode is [bracket pair colorization](https://code.visualstudio.com/blogs/2021/09/29/bracket-pair-colorization).
It looks like the `property` token counts as an opening bracket that, if used inside an `assert` or `cover` statement. However, in these cases, there is no `endproperty` token to balance the opening token.

This leads to disgracefully bracket colorization:
[image]

**Environment (please complete the following information):**
- OS: Linux x64 3.10.0-1160.71.1.el7.x86_64
- VS Code version: 1.73.1
- Extension version: 1.5.12

**Steps to reproduce**
Steps to reproduce the behavior:
- Make sure bracket pair colorization is enabled (default)
- Write some sv file with multiple `assert property` and some parenthesized expressions, like above screenshot. For example:
   ```sv
   a_pri_0:    assert property (@(posedge clk) $onehot({pri_red_0,pri_yel_0,pri_grn_0}) );
   a_pri_1:    assert property (@(posedge clk) $onehot({pri_red_1,pri_yel_1,pri_grn_1}) );
   ```

**Expected behavior**
The brackets on the two similar consecutive lines should have the same color:
[image]

**Actual behavior**
The brackets on the two similar consecutive lines have different color:
[image]

mshr-h, 2022-12-23:
That's definitely a bug!

I think the only solution is to remove the `property` `endproperty` pair from the bracket definition [here](https://github.com/mshr-h/vscode-verilog-hdl-support/blob/main/configs/systemverilog.configuration.json#L30).
I'm not an active SystemVerilog user, so I don't know if doing that is ok.

What do you think? @djoffe @Raamakrishnan

Raamakrishnan, 2022-12-23:
Ya, I feel the same. As @djoffe said, `property` can have a closing `endproperty` or not, depending on the way it is used. And I don't see a way to make the brackets optional in vscode language definitions.

mshr-h, 2022-12-26:
We have 2 options.
1. remove the `property...endproperty` pair from the bracket definition.
  - bracket colorization will work. but for example, we can't jump `property` <-> `endproperty`.
3. do nothing.
  - Opposite to option 1

I've checked some code on [chipsalliance](https://github.com/chipsalliance).
Most of the `property...endproperty` has few lines of body.
In my opinion, option 1 is better.

Raamakrishnan, 2023-03-25:
Option 3:
Add a `colorizedBracketPairs` tag in systemverilog.configuration.json. Add all the brackets, `begin..end` and others, except `property...endproperty`.
Bracket colorization will work for all brackets except for `property...endproperty`. But you can still jump between `property` <-> `endproperty`

I can't find any docs about this tag in language definition file, but there is a `editor.language.colorizedBracketPairs` [info here](https://code.visualstudio.com/updates/v1_62#_customizable-bracket-pairs)

I tried this tag in language definition file and it works. Do you want me to send a PR?

mshr-h, 2023-03-26:
@Raamakrishnan 
Sounds good to me.
Can you make the PR?

DeflateAwning, 2024-03-14:
There's no way to specify with a regex or similar that sometimes "brackets" are used for things other than pairs? For example, in Python, consider this example:

```python
a = 1
print(f"These curly braces are start/end brackets: {a}")
print("But this random curly brace { does not mess with the upcoming curly braces.")
print("Here's a again, with more curly braces that highlight properly: {a}")
```

This is still an issue; almost opened a duplicate issue. Here's yet another example:

[image]