mshr-h/vscode-verilog-hdl-support #359: Support slang

mshr-h, 2022-12-16:
[MikePopoloski/slang: SystemVerilog compiler and language services](https://github.com/MikePopoloski/slang)

retar-kamuy, 2022-12-29:
well... i'm good at b english is not.
Please tell me the isseue (purpose of this enhancement).
maybe i can help you. Perhaps I can help you, but you have to solve various problems.

retar-kamuy, 2022-12-29:
Sorry for adding questions. this is a supplement for lint check of engine (=slang).
Isn't Verilator enough? Because Verilator will support testbench Syntax in the next version (= 5.X), so I don't think slang is necessary.

mshr-h, 2023-01-06:
@retar-kamuy 
Thanks for your question.
My initial idea was to just add support linting through the slang command.
In the official doc, there's [`--lint-only`](https://sv-lang.com/command-line-ref.html) option.
I'm not familiar with Verilator. But some people want to use slang, others maybe want to use old Verilator.

retar-kamuy, 2023-01-07:
Hi, @mshr-h san
Thanks for answer.
I made concept of slang linter in [my repository](https://github.com/retar-kamuy/vscode-verilog-hdl-support/tree/feature-add-linter-slang).
Please try the linter  and let me know what you think. How to use is described on [this page](https://github.com/retar-kamuy/vscode-verilog-hdl-support/blob/feature-add-linter-slang/src/linter/README.md).

Although Slang has better linting performance than verilator, it also has the following problems.
* Installation is troublesome because Slang can not "apt-get install", Therefore, the number of people who use it will be limited.
* Too many similar configuration parameters in verilator and slang. Is it possible to merge? For example, "useWSL" parameter (Yet apples have only one parameter "verilog.linting.path". The installation path varies for each tool).

mshr-h, 2023-01-08:
@retar-kamuy 
Thanks!
I'll check it in the week.

mshr-h, 2023-01-11:
@retar-kamuy 
I've looked at the code and it's good to me.
Can you make the PR?
There's room for improvements but let's fix them in the PR branch.

> Installation is troublesome because Slang can not "apt-get install", Therefore, the number of people who use it will be limited.

I understand. It's ok.

> Too many similar configuration parameters in verilator and slang. Is it possible to merge? For example, "useWSL" parameter (Yet apples have only one parameter "verilog.linting.path". The installation path varies for each tool).

You don't have to merge it. I'll keep them in a new issue and fix it in another PR.
I may have to refactor the configuration namespace.

retar-kamuy, 2023-01-11:
Hi, @mshr-h  san
I made the PR (#386).

> You don't have to merge it. I'll keep them in a new issue and fix it in another PR.
> I may have to refactor the configuration namespace.

I  understand. But, I will not create another PR. I also want to think of a better the configuration.

mshr-h, 2023-01-11:
We have related issue here #346.

mshr-h, 2023-02-02:
Fixed by #387