mshr-h/vscode-verilog-hdl-support #360: Support rust_hdl

mshr-h, 2022-12-16:
[VHDL-LS/rust_hdl](https://github.com/VHDL-LS/rust_hdl)

zoomer-k, 2023-07-05:
@mshr-h 
try it

https://github.com/zoomer-k/vscode-verilog-hdl-support/tree/add_rust_HDL

can check syntax　

[image]

mshr-h, 2023-07-15:
@zoomer-k 
Thanks! Seems good to me. Can you make a PR if possible?

zoomer-k, 2023-07-28:
@mshr-h 
should I change readme.md?
```
{
    "[typescript]": {
        "editor.defaultFormatter": "esbenp.prettier-vscode"
    },
    "editor.formatOnSave": true,
    "editor.codeActionsOnSave": {
        "source.fixAll.eslint": true
    }
}
```

when save markdwonfile,  prettier auto format it.

mshr-h, 2023-08-04:
@zoomer-k 
Yeah.

closed 2023-08-31 by mshr-h