mshr-h/vscode-verilog-hdl-support #361: [BUG] Ctags Integration Not Working on Ubuntu 22.04

NotCamelCase, 2022-12-16:
**Describe the bug**
Despite having ctags installed and on PATH, autocomplete/outline/hover support doesn't work on Ubuntu 22.04.
I had the same issue on Ubuntu 20.04

Please note that it's started and stopped working again around the changes below, for reasons unknown to me.
Let me know if more info is required.

[1.5.9] - 2022-11-12 <- **Stopped working**
Revert [#331](https://github.com/mshr-h/vscode-verilog-hdl-support/issues/331) because instantiate module function was broken.
[1.5.8] - 2022-11-12
Fix [#332](https://github.com/mshr-h/vscode-verilog-hdl-support/issues/332)
[1.5.7] - 2022-11-11 <- **Started working**
Add toggle option for Language Server in config.
Add binary path option for Language Server in config.
Add selector option for Language Server in config.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - VS Code version 1.74.1
 - Extension version 1.5.13
 - color themes, lint tools, etc.

**Steps to reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Log**
[info] mshr-h.veriloghdl v1.5.13
[info] verilator lint requested
> verilator -sv --lint-only -I"/home/tayfun/Documents/DEV/Accellant_Private/hardware" -Wall "/home/tayfun/Documents/DEV/Accellant_Private/hardware/instruction_decode.sv"
[info] Activation complete
[info] 1 errors/warnings returned
[info] Symbols Requested: file:///home/tayfun/Documents/DEV/Accellant_Private/hardware/instruction_decode.sv
> ctags -f - --fields=+K --sort=no --excmd=n "/home/tayfun/Documents/DEV/Accellant_Private/hardware/instruction_decode.sv"
[warn] 0 top-level symbols returned
[info] verilator lint requested
> verilator -sv --lint-only -I"/home/tayfun/Documents/DEV/Accellant_Private/hardware" -Wall "/home/tayfun/Documents/DEV/Accellant_Private/hardware/instruction_issue.sv"
[info] Symbols Requested: file:///home/tayfun/Documents/DEV/Accellant_Private/hardware/instruction_issue.sv
> ctags -f - --fields=+K --sort=no --excmd=n "/home/tayfun/Documents/DEV/Accellant_Private/hardware/instruction_issue.sv"
[warn] 0 top-level symbols returned
[info] 0 errors/warnings returned
[info] Hover requested
> ctags -f - --fields=+K --sort=no --excmd=n "/home/tayfun/Documents/DEV/Accellant_Private/hardware/instruction_issue.sv"
[warn] Hover object not found
[info] Hover requested
> ctags -f - --fields=+K --sort=no --excmd=n "/home/tayfun/Documents/DEV/Accellant_Private/hardware/instruction_issue.sv"
[warn] Hover object not found
[info] Hover requested
> ctags -f - --fields=+K --sort=no --excmd=n "/home/tayfun/Documents/DEV/Accellant_Private/hardware/instruction_issue.sv"
[warn] Hover object not found
[info] Hover requested
> ctags -f - --fields=+K --sort=no --excmd=n "/home/tayfun/Documents/DEV/Accellant_Private/hardware/instruction_issue.sv"
[warn] Hover object not found

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
No info on hover/no outline, etc.

**Additional context**
Re-linting doesn't help. Module instantiation seems OK.

mshr-h, 2022-12-19:
It's working on my Ubuntu 22.04 machine which only has the verilog extension in VSCode.
My entire config looks like this.
```json
{
    "workbench.startupEditor": "none",
    "verilog.ctags.path": "/snap/bin/ctags",
    "verilog.linting.linter": "iverilog",
    "verilog.linting.path": "/usr/bin/iverilog"
}
```

Can you try to clean all the configs in VSCode? Or removing the config related to this extension is enough maybe.
The v1.5.7 had a config bug so I guess it caused the config to break.

NotCamelCase, 2022-12-19:
I found my problem. I'd set the Ctags path to /usr/bin/ctags (as that's what _which ctags_ gives me), but changing it to /snap/bin/ctags has fixed it. No idea why that update made things differently.

Thank you!

closed 2022-12-19 by NotCamelCase