mshr-h/vscode-verilog-hdl-support #376: [BUG] Command `Instantiate Module` not working

Jimmy-Yixiao, 2022-12-26:
**Describe the bug**
Nothing shows up in my current file after I enter command `Instantiate Module` and select target module.
The error in runtime status shows:
> Cannot read properties of undefined (reading 'info')
This problem is solved after I switched back to v1.5.4.

**Environment (please complete the following information):**
 - OS: [Windows 10 21H2 build 19044.2364]
 - VS Code version [1.74.2]
 - Extension version [1.8.0]
 - Universal Ctags version [6.0.0]

**Expected behavior**
Module successfully instantiated.

**Actual behavior**
Module not instantiated.

mshr-h, 2022-12-26:
Seems like newly introduced logging mechanism (#374)  is broken.
Will fix it.

mshr-h, 2022-12-26:
@Jimmy-Yixiao 
Fixed on v1.8.1.
I've uploaded the package so it'll be available soon.
Please check it out.

Jimmy-Yixiao, 2022-12-27:
@mshr-h 
Problem solved. Thank you so much. 😄

PR #377 (2022-12-26): Fix Instantiate Module
Fix #376
closed 2022-12-27 by mshr-h