mshr-h/vscode-verilog-hdl-support #397: Import of package not possible[BUG]

stu-e, 2023-02-13:
Where file B imports a package from file A there is no way to tell the lint tool where to find the package file A, then compile file A before compiling file B.

Example:

File mystuff.sv
```
package mystuff;
    localparam MYDEF = 1234;
endpackage
```

File mygadget.sv
```
import mystuff::*;
module mygadget (
    output logic [7:0] constant
);
assign constant = 8'(MYDEF);
endmodule

```

The lint tool will always highlight the import statement because it is undeclared.
This is a problem because every file that imports a package is highlighted in red making it impossible to spot files that have syntax errors.

code-tangent, 2023-02-20:
Having the same issue. How can I add packages to my linting path? Im using vivado as my linter.

AndrewNolte, 2024-04-24:
@stu-e @code-tangent This is resolved on https://github.com/AndrewNolte/vscode-system-verilog. If the package name matches the file name, it'll pick it up in the file index. But if it's a single project you could also pass it to `verilog.lint.xvlog.args`