mshr-h/vscode-verilog-hdl-support #399: [BUG]new input text will be selected when input "begin" and enter

lizhirui, 2023-02-15:
**Describe the bug**
If I input "always_ff @(posedge clk) begin" in module, and then press "enter", the code will be as below:

always_ff @(posedge clk) begin
    <-new input position
end

Then, I input some new code text, like this:
always_ff @(posedge clk) begin
    a <= 1;
end

here "a <= 1;" will be selected status.

And if I input a nested block like this:
always_ff @(posedge clk) begin
    if(b == 2) begin
<-new input position
    end
end

here, from "if" to "end" before last "end" will be all selected, and the line where new input position is  hasn't indent. Then If I want to insert a indent with TAB key, cursor will jump to after the "end" before last "end".

**Environment (please complete the following information):**
 - OS: [e.g. Windows 10 1809 build 17763.379] macOS Monterey 12.5
 - VS Code version [e.g. 1.0.0] January 2023(version 1.75)
 - Extension version [e.g. 1.0.1] v1.11.0
 - color themes, lint tools, etc.  (Configure)SystemVerilog2012 and Verilog-2005, Linter: xvlog

**Steps to reproduce**
Steps to reproduce the behavior:
1. ready a code below and save to file "test.sv":
module test;

endmodule

2. move the cursor position as below:
module test;
<-here
endmodule

3. press TAB and input "logic b;" from left to right and then press "ENTER" key twice.

3. input "always_ff @(posedge clk) begin" from left to right and then press "ENTER" key
4. now code will become as below:

module test;
    logic b;

    always_ff @(posedge clk) begin
        <-now cursor is here
    end
endmodule

5. input "if(b == 2) begin" from left to right and press "ENTER", code will become as below:
module test;
    logic b;

    always_ff @(posedge clk) begin
        if(b == 2) begin
        <- cursor is here
    end
endmodule

**BUG-1: "if(b == 2) begin" is selected**
**BUG-2: "end" isn't inputed automatically**

6. press "ENTER" again and input "end"
7. now move cursor by press "up" arrow key as below:
module test;
    logic b;

    always_ff @(posedge clk) begin
        if(b == 2) begin
<- cursor is here
        end
    end
endmodule

8. press "TAB" to insert an indent
**BUG-3: cursor will jump after "end" not insert an indent**

**Log**
Attach the log generated while following the above steps

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

mshr-h, 2023-02-15:
> BUG-1: "if(b == 2) begin" is selected
BUG-2: "end" isn't inputed automatically

You need to select begin/end *snippet*. Not the "begin" keyword which the VSCode collected from workspace files.
[image]

lizhirui, 2023-02-17:
> > BUG-1: "if(b == 2) begin" is selected
> > BUG-2: "end" isn't inputed automatically
> 
> You need to select begin/end _snippet_. Not the "begin" keyword which the VSCode collected from workspace files. [image]

OK, BUG-2 is ok, but BUG-1 is still a problem