mshr-h/vscode-verilog-hdl-support #407: iverilog linter does not work on macos with vscode 1.76.2

devindang, 2023-03-11:
**Describe the bug**
iverilog linter does not work on macos with vscode 1.76.2

**Environment (please complete the following information):**
 - OS: macOS 13.2.1
 - VS Code version: 1.76.2
 - Extension version: I didn't see ant version information but the newest
 - color themes, lint tools, etc.
   -- color themes: GitHub dark corlorblind
  -- lint tools: iverilog, 11.0

**Steps to reproduce**
Steps to reproduce the behavior:
1. Well installed iverilog and added it to envirionment variable.
2. Run "iverilog" command both in macOS terminal and vscode terminal to make sure it effected.
3. Go to vscode-verilog-hdl-support extension settings.
4. Configure verilog>linting>linter to "iverilog".
5. Configure verilog>linting>path to "iverilog".
6. And the other settings keep defaults.
7. Modified .v file to wrong syntax, and save file with command+s.
8. Nothing happened. See the error.

**Log**
Attach the log generated while following the above steps

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

devindang, 2023-03-15:
I found the issue.
If iverilog is already in PATH, DO NOT set verilog.linting.linter, after clear verilog.linting.linter, it get worked.
Also fixed the problem at windows.

Sen-Yao, 2023-03-16:
> 

I have the same issue, I tried to clear "verilog.linting.linter" but it still doesn't works. Can you please give me any other possible solutions?

devindang, 2023-03-16:
> > 
> 
> I have the same issue, I tried to clear "verilog.linting.linter" but it still doesn't works. Can you please give me any other possible solutions?

Make sure you have added the installation path of iverilog to PATH if you are using windows.

Sen-Yao, 2023-03-16:
I'm using macOS. And I have set "Verilog > Linting: Path" to "iverilog". Is that the right PATH ?

> > > 
> > 
> > 
> > I have the same issue, I tried to clear "verilog.linting.linter" but it still doesn't works. Can you please give me any other possible solutions?
> 
> Make sure you have added the installation path of iverilog to PATH if you are using windows.

I'm using macOS. And I have set "Verilog > Linting: Path" to "iverilog". Is that the right PATH ?

devindang, 2023-03-16:
> I'm using macOS. And I have set "Verilog > Linting: Path" to "iverilog". Is that the right PATH ?

Sorry, I made a mistake, it's clear "verilog.lingting.path" instead of "verilog.linting.linter"

Sen-Yao, 2023-03-16:
草 兄弟直接用中文吧……我现在 verilog.lingting.path 内是空的，然后"verilog.linting.linter"是iverilog，是这样配置的吗

Sen-Yao, 2023-03-16:
> Sorry, I made a mistake, it's clear "verilog.lingting.path" instead of "verilog.linting.linter"

草，你也是中国大学生，直接用中文吧，我现在"verilog.lingting.path" 是空的，然后 "verilog.linting.linter" 是 "iverilog"，是这样配置的吗

devindang, 2023-03-16:
> 草，你也是中国大学生，直接用中文吧，我现在"verilog.lingting.path" 是空的，然后 "verilog.linting.linter" 是 "iverilog"，是这样配置的吗

我的mac也不行，，同样的配置，windows就修复了，我以为mac也可以。怎样配置路径都不可以

Sen-Yao, 2023-03-16:
> 我的mac也不行，，同样的配置，windows就修复了，我以为mac也可以。怎样配置路径都不可以

谢谢兄弟，学电信的用macOS太折磨了 :( 我用英文反馈一下 bug 吧。

Sen-Yao, 2023-03-16:
Unfortunately, linter still doesn't work on macOS. No matter what configuration is. According to @ @devindang , Windows users can try the solution he said. But linter on macOS still doesn't work.

I guess there is a bug for macOS :(

devindang, 2023-03-17:
> Unfortunately, linter still doesn't work on macOS. No matter what configuration is. According to @ @devindang , Windows users can try the solution he said. But linter on macOS still doesn't work.
> 
> I guess there is a bug for macOS :(

use Teros HDL instead. The linting works fine with Icarus Verilog.

mshr-h, 2023-03-17:
`verilog.lingting.path` is a path to a directory where`iverilog` binary is located.
If your `iverilog` binary is `/opt/homebrew/bin/iverilog`, then you have to set `verilog.lingting.path` to `/opt/homebrew/bin`.

Maybe we need to change the description text of `verilog.lingting.path` to be more clear.

devindang, 2023-03-18:
> `verilog.lingting.path` is a path to a directory where `iverilog` binary is located. If your `iverilog` binary is `/opt/homebrew/bin/iverilog`, then you have to set `verilog.lingting.path` to `/opt/homebrew/bin`. If `iverilog` is in the PATH, you don't need to set `verilog.lingting.path`. It can be empty.
> 
> Maybe we need to change the description text of `verilog.lingting.path` to be more clear.

Is the following configuration correct?
1. verilog.linting.linter -> iverilog
2. verilog.linting.path -> /opt/homebrew/Cellar/icarus-verilog/11.0/bin/
The other options keep default. In that path, there is an "exec" file named "iverilog".

Even I configured CTAGS paths, it does still not work.
But if I change to use Teros HDL, I don't need to configure any PATHs(iverilog is already in PATH, but the extension we are taking about seems ignored it).

devindang, 2023-03-18:
> `verilog.lingting.path` is a path to a directory where `iverilog` binary is located. If your `iverilog` binary is `/opt/homebrew/bin/iverilog`, then you have to set `verilog.lingting.path` to [...]

Teros HDL extension is too bloated, I don't want to use it. So could you please help me to configure it correctly? thanks very much.

mshr-h, 2023-03-22:
What do you see in the Output Window when you execute the below process?
1. Open `language_examples/systemverilog/property_1.sv` in this repo. The file has 2 warnings.
2. Open the Output Window (You can open it by typing Ctrl+K, Ctrl+H)
3. Run `Verilog: Rerun lint tool->iverilog`

For my environment, I can see
```
2023-03-22 17:52:32.837 [info] [iverilog-lint] Execute
2023-03-22 17:52:32.837 [info] [iverilog-lint]   command: iverilog -t null -g2012  /home/ubuntu/workspace/mshr-h/vscode-verilog-hdl-support/language_examples/systemverilog/property_1.sv
2023-03-22 17:52:32.837 [info] [iverilog-lint]   cwd    : /home/ubuntu/workspace/mshr-h/vscode-verilog-hdl-support
2023-03-22 17:52:32.852 [info] [iverilog-lint] 2 errors/warnings returned
```

mshr-h, 2023-03-22:
> Is the following configuration correct?
> 
> 1. verilog.linting.linter -> iverilog
> 2. verilog.linting.path -> /opt/homebrew/Cellar/icarus-verilog/11.0/bin/
>    The other options keep default. In that path, there is an "exec" file named "iverilog".

I think it's correct.

mshr-h, 2023-03-22:
- Do you use Apple Silicon's icarus-verilog?
- What do you get when you run `which iverilog`?

devindang, 2023-03-22:
> * Do you use Apple Silicon's icarus-verilog?
> * What do you get when you run `which iverilog` in the terminal?

> I installed icarus-verilog by `brew install icarus-verilog`, and I browsed icarus-verilog repo just now, there seems no special release for apple silicon.
> When I run `which iverilog`, I get `/opt/homebrew/bin/iverilog`, it's not my path for iverilog, and I browsed that path, it is a link to '/opt/homebrew/Cellar/icarus-verilog/11.0/bin/iverilog', which is a unix executable file.

devindang, 2023-03-22:
> * Do you use Apple Silicon's icarus-verilog?
> * What do you get when you run `which iverilog` in the terminal?

And I tried to set verilog.linting.path to `/opt/homebrew/Cellar/icarus-verilog/11.0/bin/iverilog` or `/opt/homebrew/Cellar/icarus-verilog/11.0/bin/`, it still doesn't work.

devindang, 2023-03-22:
> What do you see in the Output Window when you execute the below process?
>
> 1. Open `language_examples/systemverilog/property_1.sv` in this repo. The file has 2 warnings.
> 2. Open the Output Window [...]

I didn't find "Output Window" where I can input something, but when I open command panel by `Command+Shift+P`, and run `Verilog: return lint tool`, I got an error: 

> Command "Verilog: Return lint tool" cause an error
> command 'verilog.lint' not found

mshr-h, 2023-03-22:
> * I installed icarus-verilog by `brew install icarus-verilog`, and I browsed icarus-verilog repo just now, there seems no special release for apple silicon.
> * When I run `which iverilog`, I get [...]

Both are same as my mac.

devindang, 2023-03-22:
> You can see Output Window like this. It's in the same pane as the integrated terminal. [image]
> 
> > Command "Verilog: Return lint tool" cause an error
> > command 'verilog.lint' not found
> 
> haven't seen similar error before. guess you misconfigured something. can you share your entire config file if possible?

I have tried this in my windows and got a similar output as yours. 

It asked me to choose a lint tool after executing `Verilog: return lint tool`, but I just got an error in my mac.

Besides, where can I find the config file? Do you mean the config file of the Verilog extension?

mshr-h, 2023-03-22:
> Besides, where can I find the config file? Do you mean the config file of the Verilog extension?

Open the settings tab and click "Open Settings (JSON)" in the top right corner.
[image]

devindang, 2023-03-22:
> This is my settings json
> 
> ```
> {
>     "workbench.preferredDarkColorTheme": "GitHub Dark Colorblind (Beta)",
>     "workbench.preferredLightColorTheme": "GitHub Light Colorblind (Beta)",
>     "git.openRepositoryInParentFolders": "always",
>     "verilog.linting.linter": "iverilog",
>     "php.validate.run": "onType",
>     // "workbench.colorCustomizations": {
>     //     "editor.selectionBackground": "#898900",
>     //     "editor.selectionHighlightBackground": "#c5293e"
>     // },
>     "workbench.iconTheme": "vscode-icons-mac",
>     "matlab.mlintpath": "/Applications/MATLAB_R2022b.app/bin/maci64/mlint",
>     "verilog.ctags.path": "",
>     "verilog.linting.path": "/opt/homebrew/Cellar/icarus-verilog/11.0/bin/",
>     "workbench.colorTheme": "GitHub Dark Colorblind (Beta)",
>     "window.autoDetectColorScheme": true
> }
> ```

This is my settings json

```
{
    "workbench.preferredDarkColorTheme": "GitHub Dark Colorblind (Beta)",
    "workbench.preferredLightColorTheme": "GitHub Light Colorblind (Beta)",
    "git.openRepositoryInParentFolders": "always",
    "verilog.linting.linter": "iverilog",
    "php.validate.run": "onType",
    // "workbench.colorCustomizations": {
    //     "editor.selectionBackground": "#898900",
    //     "editor.selectionHighlightBackground": "#c5293e"
    // },
    "workbench.iconTheme": "vscode-icons-mac",
    "matlab.mlintpath": "/Applications/MATLAB_R2022b.app/bin/maci64/mlint",
    "verilog.ctags.path": "",
    "verilog.linting.path": "/opt/homebrew/Cellar/icarus-verilog/11.0/bin/",
    "workbench.colorTheme": "GitHub Dark Colorblind (Beta)",
    "window.autoDetectColorScheme": true
}
```

mshr-h, 2023-03-23:
seems ok to me.
I've applied your Verilog config to my mac and it works correctly.
I have no idea...

devindang, 2023-03-23:
> seems ok to me. I've applied your Verilog config to my mac and it works correctly. I have no idea...

Can you find the reason for the following behavior? What does 'verilog.lint' refer to?
> run Verilog: return lint tool, I got an error:
> `Command "Verilog: Return lint tool" cause an error`
> `command 'verilog.lint' not found`

mshr-h, 2023-03-23:
> Can you find the reason for the following behavior? What does 'verilog.lint' refer to?

The `Verilog: Rerun lint tool` command is defined [here](https://github.com/mshr-h/vscode-verilog-hdl-support/blob/main/package.json#L497). And it invokes `verilog.lint` function defined [here](https://github.com/mshr-h/vscode-verilog-hdl-support/blob/main/src/extension.ts#L164).

I guess the extension initialization process has some problem.

Did you try to reinstall the extension?

devindang, 2023-03-23:
> seems ok to me. I've applied your Verilog config to my mac and it works correctly. I have no idea...

Can you find the reason of the following behavior? What does 'verilog.lint' refer to?
> run Verilog: return lint tool, I got an error:
> `Command "Verilog: Return lint tool" cause an error`
> `command 'verilog.lint' not found`

> > Can you find the reason for the following behavior? What does 'verilog.lint' refer to?
> 
> I couldn't find why it happens. The `Verilog: Rerun lint tool` command is defined [here](https://github.com/mshr-h/vscode-verilog-hdl-support/blob/main/package.json#L497). And it invokes `verilog.lint` function defined [here](https://github.com/mshr-h/vscode-verilog-hdl-support/blob/main/src/extension.ts#L164).
> 
> I guess the extension initialization process has some problem.
> 
> Did you try to reinstall the extension?

Yes, I have tried to reinstall the extension, but it did not work.

mshr-h, 2023-03-23:
I can't think of any good ideas for investigating.
Sorry for that...

devindang, 2023-03-23:
> I can't think of any good ideas for investigating.
> Sorry for that...

Don't feel sorry. I tried to reinstall the extension, now it can execute "Verilog: Return lint tool", and ask me to choose a linter tool. 

If I choose "iverilog", I get an error:

[image]

Does this tip help you find the cause of the issue?

mshr-h, 2023-03-24:
Do you see any text in the Output Window when you execute "Verilog: Return lint tool"?

devindang, 2023-03-24:
No, there is no any output in the "Output" windows, just a pop-up error tips, even I change the output source to "Verilog"

mshr-h, 2023-03-24:
What do you see when you execute "Verilog: Rerun lint tool" on a non-Verilog/SystemVerilog file?
For my mac, I see the error "Verilog-HDL/SystemVerilog: No document opened".

mshr-h, 2023-03-24:
> What do you see when you execute "Verilog: Rerun lint tool" on a non-Verilog/SystemVerilog file?
> For my mac, I see the error "Verilog-HDL/SystemVerilog: No document opened".

Ah, never mind. You must see "Verilog-HDL/SystemVerilog: No document opened".

mshr-h, 2023-03-24:
The error happens between [LintManager.ts#L128](https://github.com/mshr-h/vscode-verilog-hdl-support/blob/main/src/linter/LintManager.ts#L128) and [IcarusLinter.ts#L60](https://github.com/mshr-h/vscode-verilog-hdl-support/blob/main/src/linter/IcarusLinter.ts#L60).
I'll add some logging messages.

devindang, 2023-03-24:
> > What do you see when you execute "Verilog: Rerun lint tool" on a non-Verilog/SystemVerilog file?
> > For my mac, I see the error "Verilog-HDL/SystemVerilog: No document opened".
> 
> Ah, never mind. You must see "Verilog-HDL/SystemVerilog: No document opened".

Yes, I saw "Verilog-HDL/SystemVerilog: No document opened".

devindang, 2023-03-24:
> The error happens between [LintManager.ts#L128](https://github.com/mshr-h/vscode-verilog-hdl-support/blob/main/src/linter/LintManager.ts#L128) and [IcarusLinter.ts#L60](https://github.com/mshr-h/vscode-verilog-hdl-support/blob/main/src/linter/IcarusLinter.ts#L60).
> I'll add some logging messages.

Well, thanks a lot

mshr-h, 2023-03-24:
Released v1.11.2 with more logging messages.
Please try it and share Output Window when you execute "Verilog: Rerun lint tool".

For my mac,

devindang, 2023-03-24:
> Released v1.11.2 with more logging messages. Please try it and share Output Window when you execute "Verilog: Rerun lint tool".
> 
> For my mac, I got this
> 
> > 2023-03-24 18:29:02.117 [info] [lint-manager] Executing LintManagerrunLintTool()
> > 2023-03-24 18:29:02.118 [info] [lint-manager] document.languageId = systemverilog
> > 2023-03-24 18:29:02.314 [info] [lint-manager] linterStr = {"label":"iverilog","description":"Icarus Verilog"}
> > 2023-03-24 18:29:02.315 [info] [lint-manager] linter = {"diagnosticCollection":{"d":"_generated_diagnostic_collection_name_#4","e":"_generated_diagnostic_collection_name_#4","f":1000,"c":false},"name":"iverilog","logger":{"name":"Verilog","logLevel":3},"linterInstalledPath":"","configuration":{"arguments":"","includePath":[],"verilogHDL":{"standard":"Verilog-2005"},"systemVerilog":{"standard":"SystemVerilog2012"},"runAtFileLocation":false},"arguments":"","includePath":[],"standards":{},"runAtFileLocation":false}
> > 2023-03-24 18:29:02.315 [info] [lint-manager] vscode.window.activeTextEditor.document = {"uri":{"$mid":1,"fsPath":"/Users/mshr-h/workspace/mshr-h/vscode-verilog-hdl-support/language_examples/systemverilog/property_1.sv","external":"file:///Users/mshr-h/workspace/mshr-h/vscode-verilog-hdl-support/language_examples/systemverilog/property_1.sv","path":"/Users/mshr-h/workspace/mshr-h/vscode-verilog-hdl-support/language_examples/systemverilog/property_1.sv","scheme":"file"},"fileName":"/Users/mshr-h/workspace/mshr-h/vscode-verilog-hdl-support/language_examples/systemverilog/property_1.sv","isUntitled":false,"languageId":"systemverilog","version":1,"isClosed":false,"isDirty":false,"eol":1,"lineCount":7}
> > 2023-03-24 18:29:02.315 [info] [iverilog-lint] Executing IcarusLinter.lint()
> > 2023-03-24 18:29:02.315 [info] [iverilog-lint] iverilog binary path: iverilog
> > 2023-03-24 18:29:02.315 [info] [iverilog-lint] Execute
> > 2023-03-24 18:29:02.315 [info] [iverilog-lint]   command: iverilog -t null -g2012  /Users/mshr-h/workspace/mshr-h/vscode-verilog-hdl-support/language_examples/systemverilog/property_1.sv
> > 2023-03-24 18:29:02.315 [info] [iverilog-lint]   cwd    : /Users/mshr-h/workspace/mshr-h/vscode-verilog-hdl-support
> > 2023-03-24 18:29:02.334 [info] [iverilog-lint] 2 errors/warnings returned

I got this, and see the picture

> 2023-03-24 18:23:52.008 [info] Hover requested
> 2023-03-24 18:23:54.224 [info] [lint-manager] Executing LintManagerrunLintTool()
> 2023-03-24 18:23:54.224 [info] [lint-manager] document.languageId = verilog
> 2023-03-24 18:23:56.298 [info] [lint-manager] linterStr = {"label":"iverilog","description":"Icarus Verilog"}
> 2023-03-24 18:23:56.298 [info] [lint-manager] linter = {"diagnosticCollection":{"d":"_generated_diagnostic_collection_name_#0","e":"_generated_diagnostic_collection_name_#0","f":1000,"c":false},"name":"iverilog","logger":{"name":"Verilog","logLevel":3},"linterInstalledPath":"/opt/homebrew/Cellar/icarus-verilog/11.0/bin/","configuration":{"arguments":"","includePath":[],"verilogHDL":{"standard":"Verilog-2005"},"systemVerilog":{"standard":"SystemVerilog2012"},"runAtFileLocation":false},"arguments":"","includePath":[],"standards":{},"runAtFileLocation":false}
> 2023-03-24 18:23:56.298 [info] [lint-manager] vscode.window.activeTextEditor.document = {"uri":{"$mid":1,"fsPath":"/Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v","external":"file:///Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v","path":"/Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v","scheme":"file"},"fileName":"/Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v","isUntitled":false,"languageId":"verilog","version":1,"isClosed":false,"isDirty":false,"eol":1,"lineCount":189}
> 2023-03-24 18:23:56.298 [info] [iverilog-lint] Executing IcarusLinter.lint()
> 2023-03-24 18:23:56.298 [info] [iverilog-lint] iverilog binary path: /opt/homebrew/Cellar/icarus-verilog/11.0/bin/iverilog

[image]

mshr-h, 2023-03-24:
Thanks.
Can you try with [this extension](https://drive.google.com/file/d/1mi0g3YSD77Ypcsl-USbvOLblLVL2OtYx/view?usp=share_link).
I added more logging messages. It's not officially released.
The source code is at https://github.com/mshr-h/vscode-verilog-hdl-support/tree/debug-407.

devindang, 2023-03-24:
> Thanks.
> Can you try with [this extension](https://drive.google.com/file/d/1mi0g3YSD77Ypcsl-USbvOLblLVL2OtYx/view?usp=share_link).
> I added more logging messages. It's not officially released.
> The source code is at https://github.com/mshr-h/vscode-verilog-hdl-support/tree/debug-407.

Yes, the following is the result.

> 2023-03-24 19:14:26.184 [info] mshr-h.veriloghdl is now active.
> 2023-03-24 19:14:26.184 [info] [Ctags] ctags manager configure
> 2023-03-24 19:14:26.184 [info] [lint-manager] Using linter: iverilog
> 2023-03-24 19:14:26.184 [info] [iverilog-lint] Executing IcarusLinter.lint()
> 2023-03-24 19:14:26.184 [info] [iverilog-lint] iverilog binary path: iverilog
> 2023-03-24 19:14:26.184 [info] [iverilog-lint] doc.languageId = verilog
> 2023-03-24 19:14:26.184 [info] [iverilog-lint] includePath = []
> 2023-03-24 19:14:26.184 [info] [iverilog-lint] arguments = ""
> 2023-03-24 19:14:26.184 [info] [iverilog-lint] doc.uri.fsPath = "/Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v"
> 2023-03-24 19:14:26.184 [info] [iverilog-lint] args: ["-t null","-g2005","","/Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v"]
> 2023-03-24 19:14:26.184 [info] [iverilog-lint] Execute
> 2023-03-24 19:14:26.184 [info] [iverilog-lint]   command: iverilog -t null -g2005  /Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v
> 2023-03-24 19:14:26.184 [info] [VerilogSymbol] Symbols Requested: file:///Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v
> 2023-03-24 19:14:26.184 [info] [Ctags] indexing...
> 2023-03-24 19:14:26.184 [info] [Ctags] executing ctags
> 2023-03-24 19:14:26.184 [info] [VerilogSymbol] Symbols Requested: file:///Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v
> 2023-03-24 19:14:26.184 [info] [Ctags] indexing...
> 2023-03-24 19:14:26.184 [info] [Ctags] executing ctags
> 2023-03-24 19:14:26.184 [info] [VerilogSymbol] Symbols Requested: file:///Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v
> 2023-03-24 19:14:26.184 [info] [Ctags] indexing...
> 2023-03-24 19:14:26.184 [info] [Ctags] executing ctags

[image]

mshr-h, 2023-03-25:
I see.
Can you try [this](https://drive.google.com/file/d/1ToIDFT5h_eBjFt0mOP6UQU1sbLq-DMZ3/view?usp=share_link)?
Added more log messages.

devindang, 2023-03-25:
> 

This is the outputs:

> 2023-03-25 15:30:16.282 [info] mshr-h.veriloghdl is now active.
> 2023-03-25 15:30:16.282 [info] [lint-manager] Using linter: iverilog
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] Executing IcarusLinter.lint()
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] iverilog binary path: iverilog
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] doc.languageId = verilog
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] includePath = []
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] arguments = ""
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] doc.uri.fsPath = "/Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v"
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] args: ["-t null","-g2005","","/Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v"]
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] Execute
> 2023-03-25 15:30:16.282 [info] [iverilog-lint]   command: iverilog -t null -g2005  /Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] runAtFileLocation = false
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] doc.uri = {"$mid":1,"fsPath":"/Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v","external":"file:///Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v","path":"/Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v","scheme":"file"}
> 2023-03-25 15:30:16.282 [info] [iverilog-lint] vscode.workspace.workspaceFolders = undefined
> 2023-03-25 15:30:16.282 [info] [VerilogSymbol] Symbols Requested: file:///Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v
> 2023-03-25 15:30:16.282 [info] [VerilogSymbol] Symbols Requested: file:///Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v
> 2023-03-25 15:30:16.282 [info] [VerilogSymbol] Symbols Requested: file:///Users/devin/Documents/workspace/amba/ahb2apb/ahb_to_apb.v
> 2023-03-25 15:30:31.589 [info] Hover requested
> 2023-03-25 15:30:31.785 [info] Hover requested
> 2023-03-25 15:30:34.310 [info] Hover requested

mshr-h, 2023-03-25:
I could reproduce the problem on Windows.
[image]

The error is [IcarusLinter.ts#L70).

When you don't open the workspace folder and you try to run lint, then you get the error.
If the `runAtFileLocation` option is false, then cwd (current working directory of iverilog process) will be the path of the workspace folder but you didn't open it. So cwd will be undefined.

mshr-h, 2023-03-25:
I'm not going to solve the problem soon because I don't use it in such a way. I always open the workspace folder.
But PR is welcome.

mshr-h, 2023-03-26:
The problem might be solved by enabling "runAtFileLocation" option. It forces to use the file location as the cwd.

closed 2023-03-27 by mshr-h

mshr-h, 2023-03-27:
The problem may have been fixed in v1.11.3.
You need to uninstall the extension and install it from the marketplace.

devindang, 2023-03-27:
> The problem may have been fixed in v1.11.3.
> You need to uninstall the extension and install it from the marketplace.
> If you still have the problem, please reopen the issue.

Thank you, it gets worked now when I just open a file without opening a folder.

commit f5179c35 (2023-03-27): fix #407 (#411)