mshr-h/vscode-verilog-hdl-support #413: [BUG] Error while using verible-verilog-format: "spawnSync verible-verilog-format ENOENT"

hui-shao, 2023-04-08:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04.2
 - VS Code version: 
```
版本: 1.77.1
提交: b7886d7461186a5eac768481578c1d7ca80e2d21
日期: 2023-04-04T23:20:37.202Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Linux x64 5.19.0-38-generic
沙盒化: Yes
```
 - Extension version: 1.11.3
 - Verible version: [verible-v0.0-3132-gb052efb1-Ubuntu-22.04-jammy-x86_64.tar.gz](https://github.com/chipsalliance/verible/releases/download/v0.0-3132-gb052efb1/verible-v0.0-3132-gb052efb1-Ubuntu-22.04-jammy-x86_64.tar.gz)

**Steps to reproduce**
Steps to reproduce the behavior:
1. Create a new file ending with .v and write some code.
2. In VS Code, Right click - format document

**Log**
```
2023-04-08 16:45:23.561 [info] Temp file created at:/tmp/veribleVerilogFormatter-4683fa1252d592b5049ae53934982702.tmp.v
2023-04-08 16:45:23.561 [info] Executing command: verible-verilog-format --inplace --column_limit=200 --indentation_spaces=2 /tmp/veribleVerilogFormatter-4683fa1252d592b5049ae53934982702.tmp.v
2023-04-08 16:45:23.561 [error] Error: spawnSync verible-verilog-format ENOENT
```
[image]

**Expected behavior**
The document is formatted and there is no errors.

**Actual behavior**
The document is not formatted. And In the VS Code's "Output" tab, you can see the error information in the above "Log"

**Additional context**

- I have added the format tool to PATH.
- I've tried running verible-verible-format in the Ubuntu terminal and it works and formats the file.

settings:
```
    "verilog.ctags.path": "ctags",
    "verilog.linting.linter": "xvlog",
    "verilog.formatting.verilogHDL.formatter": "verible-verilog-format",
    "verilog.formatting.veribleVerilogFormatter.arguments": "--column_limit=200 --indentation_spaces=2"
```

hui-shao, 2023-04-09:
I restarted the computer and this error disappeared.
? ? ? very strange.

closed 2023-04-09 by hui-shao

MahmoudKMaarouf, 2025-08-12:
Have a similar bug but it does not go away. It is becaues I am using a .v file not .sv file.

MahmoudKMaarouf, 2025-08-12:
Hi thanks for your feedback. I am certain there is not Verilog syntax errors, I've tested on multiple files and it only doesn't work with .v files. Thank you!