mshr-h/vscode-verilog-hdl-support #417: [BUG] "Cannot read properties of undefined (reading 'then')" error in logs repeatedly.

rossc719g, 2023-05-26:
```
2023-05-25 16:54:57.756 [error] Cannot read properties of undefined (reading 'then'): TypeError: Cannot read properties of undefined (reading 'then')
    at /Users/rossc719/.vscode/extensions/mshr-h.veriloghdl-1.11.4/dist/main.js:61:71439
    at new Promise (<anonymous>)
    at Vh.index (/Users/rossc719/.vscode/extensions/mshr-h.veriloghdl-1.11.4/dist/main.js:61:71384)
    at Function.getSymbols (/Users/rossc719/.vscode/extensions/mshr-h.veriloghdl-1.11.4/dist/main.js:61:71943)
    at P_.provideDocumentSymbols (/Users/rossc719/.vscode/extensions/mshr-h.veriloghdl-1.11.4/dist/main.js:584:5874)
    at s.provideDocumentSymbols (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:100:39971)
    at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:100:65925
    at le.s (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:100:65165)
    at le.$provideDocumentSymbols (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:100:65912)
    at a.N (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:110:11620)
    at a.M (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:110:11338)
    at a.H (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:110:10393)
    at a.G (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:110:9412)
    at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:110:8200
    at v.invoke (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:63:145)
    at h.deliver (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:63:2121)
    at i.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:63:1729)
    at g.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:72:14852)
    at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:126:32403
    at v.invoke (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:63:145)
    at h.deliver (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:63:2121)
    at i.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:63:1729)
    at g.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:72:14852)
    at MessagePortMain.<anonymous> (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:126:30529)
    at MessagePortMain.emit (node:events:513:28)
    at Object.MessagePortMain._internalPort.emit (node:electron/js2c/utility_init:2:367)
```

 - OS: MacOS Ventura 13.3.1
 - VS Code version 1.78.2
 - Extension version 1.11.4
 - color themes, lint tools, etc.: I installed all the lint tools that are supported, and ctags, etc.. 

**Steps to reproduce**
1. The error pops up for me whenever I save, edit, etc.. Just pops up continuously.

**Log**
Logs above.

**Expected behavior**
Unclear.  I am a new user, so I am not sure what is failing.  Hard to know what it is not doing.

**Actual behavior**
Lots of that error message.

**Additional context**
Cant think of anything else.. happy to run tests or debug on my end if that would help.

LnnrtS, 2023-06-06:
Do you have ctags available in your path? Had the same problem and solved it that way

rossc719g, 2023-06-06:
Wow, thanks!  Working now.

I did have ctags installed, but I did not change the ctags path from "none" to "ctags".

I guess I misunderstood what the default of "none" meant in this context.
It feels strange for it to fail in this way using the default.

Thanks again!

LnnrtS, 2023-06-06:
Great it works!

I agree it's not doing as what one would expect in this situation.

TheLapinMalin, 2024-03-10:
I've run into the same issue, but don't have ctags installed, so I've proposed a fix for this issue here: https://github.com/mshr-h/vscode-verilog-hdl-support/pull/469